hku-mars/FAST-LIVO2 #117: [ WARN] [1740646406.241370025]: IMU and LiDAR not synced! delta time: -0.691485 .

lyf106209, 2025-02-27:
I am currently running my own data with Fast-LIVO2, and regardless of whether I set img_en: 0 or img_en: 1, I receive the following warning message:

[ WARN] [1740646406.241370025]: IMU and LiDAR not synced! delta time: -0.691485.

I have not performed any synchronization between the IMU and LiDAR data, as I am using raw data from Avia. I initially assumed that the IMU and LiDAR data were already synchronized.

Could you please advise on what steps I should take to resolve this issue?  Also, any guidance on how to properly synchronize the IMU and LiDAR data would be greatly appreciated.

xuankuzcr, 2025-03-01:
To my knowledge, the Livox Avia and its built-in IMU are synchronized. I suspect the issue might be caused by insufficient data transmission bandwidth.

lyf106209, 2025-03-04:
> To my knowledge, the Livox Avia and its built-in IMU are synchronized. I suspect the issue might be caused by insufficient data transmission bandwidth.

Thank you very much for your response. After hardware synchronization, we recorded the rosbag in outdoor environment using the parameters `-b 4096 --split --size=4096`, and we have not encountered this issue again.

wolichen, 2025-03-16:
> Thank you very much for your response. After hardware synchronization, we recorded the rosbag in outdoor environment using the parameters `-b 4096 --split --size=4096`, and we have not encountered [...]

Excuse me, for the previous problem, did the screen appear sometimes, and then when there was an error, this situation occurred? How did you solve it? Thank you!

lyf106209, 2025-03-17:
> > -b 4096 --split --size=4096
> > Excuse me, could you kindly clarify which tool these parameters belong to? Are they parameters of rosbag record? Thank you!

yes, they are parameters of rosbag record.

lyf106209, 2025-03-17:
> Excuse me, for the previous problem, did the screen appear sometimes, and then when there was an error, this situation occurred? How did you solve it? Thank you!

I think this WARN is related to time synchronization and data transmission. Perhaps you could first try hardware synchronization for the sensors or perform time synchronization on the data.

closed 2025-03-17 by lyf106209

wolichen, 2025-03-24:
> > > > To my knowledge, the Livox Avia and its built-in IMU are synchronized. I suspect the issue might be caused by insufficient data transmission bandwidth.
> > > 
> > > 
> > > Thank you very much for your response. After hardware synchronization, we recorded the rosbag in outdoor environment using the parameters `-b 4096 --split --size=4096`, and we have not encountered this issue again.
> > 
> > 
> > Excuse me, for the previous problem, did the screen appear sometimes, and then when there was an error, this situation occurred? How did you solve it? Thank you!
> 
> I think this WARN is related to time synchronization and data transmission, just as mentioned above. Perhaps you could first try hardware synchronization for the sensors or perform time synchronization on the data.

Hi, thanks a lot. I’m just getting started with this, so I wanted to ask how you implemented your time hardware synchronization? After I flashed the program, I found that the system couldn’t recognize the serial port. I checked online and saw that I need to use STM32CubeMX to configure the USB CDC virtual serial port and then flash it again, but doing so would overwrite my previously flashed code, so I would need to merge the two projects before flashing. I'm not very familiar with how to do that. Later, I also found that I could use a USB-to-TTL adapter like CH340 instead. May I ask how you implemented yours? Could you please teach me? Thank you so much!