hku-mars/FAST-LIVO2 #126: Hight Position Drift Issue

crazy-robotics, 2025-02-28:
I am working on indoor UAV `SLAM` using a `Pixhawk 6C` running `PX4 v1.15.2` paired with a `Jetson Orin Nano` as the companion computer. For localization, I am using a `Livox Mid-360 LiDAR` and running [FAST-LIVO2](https://github.com/hku-mars/FAST-LIVO2) for `local position estimation`.  

After launching `FAST-LIVO2`, it publishes the estimated pose on `/mavros/vision_pose/pose`. I have configured the PX4 parameters according to the [Using Vision or Motion Capture Systems for Position Estimation](https://docs.px4.io/main/en/ros/external_position_estimation.html) guide. Initially, in `QGroundControl (QGC)`, I observe a `no valid local position `warning, but this disappears once FAST-LIVO2 starts running, indicating that the system is receiving position estimates.  

However, when switching to `Position mode`, the drone successfully takes off, but it experiences **significant drift** without any manual input. The drift is highly unstable, making it difficult to control the UAV.  

I have attached multiple test flight logs for reference:  
📌 [Test-flight1](https://review.px4.io/plot_app?log=7ee4986e-2aeb-4dba-91ea-a73b81072f68)  
📌 [Test-flight2](https://review.px4.io/plot_app?log=eef83bf5-451b-403c-b363-1aae75294bc9)  
📌 [Test-flight3](https://review.px4.io/plot_app?log=fcc6de1d-b40f-4d82-a622-22e6f934aa56)  
📌 [Test-flight4](https://review.px4.io/plot_app?log=5c6705a3-bc5f-4a07-8c2c-4f9306e182ab)  

I would appreciate any suggestions on how to resolve the `position drift issue` 

Any help or guidance would be greatly appreciated. Thanks in advance!

Es1erda, 2025-02-28:
It seems to be related to how PX4 handles position estimates. I suggest posting the issue in the PX4 related GitHub repository or the PX4 forum for better support. Based on my experience, FAST-LIVO2 doesn't typically cause hight position drift so easily.

xuankuzcr, 2025-03-01:
Perhaps you could ask the student in issue #82; it seems the results he’s achieving are quite stable.