hku-mars/FAST-LIVO2 #127: Sensor dependency

babakbch, 2025-02-28:
Can the algorithm work only with the LiDAR and IMU data? can the camera topic be removed somehow and still produce a point cloud map from LiDAR without the RGB data?

Es1erda, 2025-03-01:
Sure, you can set `img_en` to 0 in the config/avia.yaml file to disable the image input and get the LIO results.

babakbch, 2025-03-01:
Awesome! thank you

closed 2025-03-01 by babakbch

xuankuzcr, 2025-03-01:
Our algorithm is capable of operating directly in `LO, LIO, and LIVO` mode, all of which are state-of-the-art solutions.

babakbch, 2025-03-01:
> Our algorithm is capable of operating directly in `LO, LIO, and LIVO` mode, all of which are state-of-the-art solutions.

Congratulations on a great job done , can't wait to test it on different data and specially ones we gathered ourselves.