hku-mars/FAST-LIVO2 #128: ROS2 HUMBLE Migration

sujit-168, 2025-03-02:
@xuankuzcr  Dr. Zheng,  

### Topic Area
New Feature about  #40  #55 

### Feature Area
This issue concerns this project's [pull request](https://github.com/Robotic-Developer-Road/FAST-LIVO2/tree/humble) for ROS2 migration (ROS2 HUMBLE).

### Body

 If merges into this repository are allowed, I would like to have a separate humble branch for storage without affecting the updates of the main branch.

### 

Once again, salute your excellent work. Best wishes!

OxygenLost, 2025-03-03:
https://github.com/Robotic-Developer-Road/rpg_vikit.git 仓库无法访问，是私有仓库吗？

xuankuzcr, 2025-03-03:
@sujit-168 Thank you for your effort and support regarding the ROS2 Humble migration. Feel free to submit a pull request if it’s convenient for you. I’m currently occupied with other commitments and won’t be able to address this immediately, but I’ll review and process it as soon as I have the time.

Thank you for your patience, and best wishes!

sujit-168, 2025-03-03:
Understood. I recommend creating a humble branch based on the current main branch in this repository. This would allow me to submit pull requests targeting the humble branch instead of merging directly into main. Such a workflow would help prevent unintended impacts on the ROS1-related code in the main branch.

sujit-168, 2025-03-03:
> https://github.com/Robotic-Developer-Road/rpg_vikit.git 仓库无法访问，是私有仓库吗？

@OxygenLost The repository in question is currently set to private, with plans to transition it to a public status in the near future.

YanhaoZhang, 2025-03-14:
Hi @sujit-168. I saw your video on bilibili. Really appreciate your work.  I was wondering if the mitigation has been done and you would consider releasing the code?

sujit-168, 2025-03-14:
> Hi @sujit-168. I saw your video on bilibili. Really appreciate your work.  I was wondering if the mitigation has been done and you would consider releasing the code? 
@YanhaoZhang I got it, thank you for your follow, you said the code is

sujit-168, 2025-03-14:
@YanhaoZhang Thank you for watching my video on Bilibili. Are you asking about the GitHub repository [rpg_vikit](https://github.com/Robotic-Developer-Road/FAST-LIVO2/tree/humble?tab=readme-ov-file#24-vikit) mentioned by OxygenLost?

YanhaoZhang, 2025-03-14:
Hi @sujit-168. Yes. Thanks a lot! I did not know you had already released the code. Really appreciate it!

sujit-168, 2025-03-14:
@YanhaoZhang Due to copyright risks, I am sorry that I may not make the rpg_vikit mentioned in project [Readme.md](https://github.com/Robotic-Developer-Road/FAST-LIVO2/tree/humble?tab=readme-ov-file#24-vikit) public shortly, unless I get permission from the original author. Thanks for understanding about this!

YanhaoZhang, 2025-03-16:
@sujit-168 Sorry for the late reply. Yes, I totally understand this, and I really appreciate you releasing the mitigation as well as your prompt reply.

Divepit, 2025-03-31:
Hello all, thank you for your work! Do I understand correctly that vikit is required and none of the provided options can be built using colcon but only with catkin?

I would like to try compatibility on ROS Jazzy, Ubuntu 24.04 but I can not use anything that is based on catkin..

sujit-168, 2025-04-07:
@Divepit Any contribution based on [PR](https://github.com/hku-mars/FAST-LIVO2/pull/131) is welcome. You can share your test results and compatibility issues about this `PR` with ROS Jazzy on Ubuntu 24.04

TannerGilbert, 2025-04-07:
@Divepit for me Fast-LIVO 2 runs on Humble with the forks from @james-ward after installing the newest `sophus` version via apt:

His forks:
- https://github.com/integralrobotics/FAST-LIVO2
- https://github.com/integralrobotics/rpg_vikit

Install Sophus
```
sudo apt install ros-$ROS_DISTRO-sophus
```

TannerGilbert, 2025-04-08:
I opened two PRs to @james-ward's fork. One to fix builds for ROS 2 versions that are older than Iron (so including Humble) and the second one to fix the Velodyne processing similar to #53 for ROS 1.

https://github.com/integralrobotics/FAST-LIVO2/pull/1
https://github.com/integralrobotics/FAST-LIVO2/pull/2

If @james-ward decides to merge these PR's they would then also be included in the PR to @sujit-168's fork: https://github.com/Robotic-Developer-Road/FAST-LIVO2/pull/1

biggiantpigeon, 2025-04-16:
After I convert sample data bag from ROS1 to ROS2, I find the lidar and IMU data is no longer synced with lidar delayed about 0.5~1 second, and some times log prints `IMU and LiDAR not synced! delta time: xxx`. Maybe something happens during conversion, but seems ok after running for a while

hujiax380, 2025-06-20:
> After I convert sample data bag from ROS1 to ROS2, I find the lidar and IMU data is no longer synced with lidar delayed about 0.5~1 second, and some times log prints `IMU and LiDAR not synced! delta time: xxx`. Maybe something happens during conversion, but seems ok after running for a while

Have you solved this problem? As the program runs for a longer time, the ''delta time'' keeps increasing. It seems that the latest LiDAR data is subscribed to, but the system is still processing IMU data from a long time ago.

HuqMBT, 2025-09-09:
I want to test it with Lite3 Robot. Can anyone help me to rebuild it with ROS2 Foxy? I got stuck on the point while colcon build it as point cloud ros library is not available in Foxy. So can anyone tell me if I change the CMakeLists.txt and remove find_package(pcl_ros REQUIRED) will it make a lot of problems? as find_package(PCL REQUIRED) is already there.

HuqMBT, 2025-09-09:
> I want to test it with Lite3 Robot. Can anyone help me to rebuild it with ROS2 Foxy? I got stuck on the point while colcon build it as point cloud ros library is not available in Foxy. So can anyone help me with changes in the CMakeLists.txt?

I am getting the error in the text file while building it for ROS2 Foxy. I have successfully installed Sophus, PCL-ROS for foxy, Vikit and livox_ros_driver2. Can Anyone help me with this to run it in Foxy? I would then try it on Lite 3 Robot.

[problem.txt](https://github.com/user-attachments/files/22234113/problem.txt)