hku-mars/FAST-LIVO2 #129: D435i+MID 360

yzp1767791249, 2025-03-03:
大佬们您们好，请问能否使用D435i和mid 360激光雷达替代实现呀，这样是不是在时间同步过程不太好处理呀，有相关实例或教程吗，十分感谢！

xuankuzcr, 2025-03-03:
可以替代，D435i的外部触发参考这个issue [#2637](https://github.com/IntelRealSense/librealsense/issues/2637)，和Mid360之间的同步参考这个仓库 https://github.com/xuankuzcr/LIV_handhold

yzp1767791249, 2025-03-03:
好滴好滴非常感谢，我最近试一下然后给您反馈！！！

autumntang, 2025-03-05:
@yzp1767791249 您好，我最近也在做mid360和d435i的相关工作，请问可以加个联系方式交流一下吗，我的qq是2319909319

xuankuzcr, 2025-03-06:
@yzp1767791249 我之前说的好像有问题，建议你再详细看下D435i的手册和相关issue #145 https://github.com/IntelRealSense/librealsense/issues/2637 https://github.com/IntelRealSense/librealsense/issues/12064

ErcBunny, 2025-03-29:
Hi，感谢开源。我想请教一下有没有可能直接用d435i输出的点云代替lidar点云呢？搜了一圈貌似没有看到类似的做法，如果不行的话具体问题出在哪儿呢？感谢～

18245934835ly, 2025-03-29:
b站上有用d435i输出的点云代替lidar点云复现成功的视频，可以参考下 https://www.bilibili.com/video/BV1eFoBY2Egg/?spm_id_from=333.337.search-card.all.click

ErcBunny, 2025-03-29:
> @ErcBunny b站上有用d435i输出的点云代替lidar点云复现成功的视频，可以参考下 https://www.bilibili.com/video/BV1eFoBY2Egg/?spm_id_from=333.337.search-card.all.click    视频好像没复现成功，好的效果在评论区～

@18245934835ly 我看视频里面描述为d435i只用于提供rgb图像。并没有替代lidar点云，疑惑🤔