hku-mars/FAST-LIVO2 #134: Normalized Cross-Correlation (NCC) seems to be disabled.

DapengFeng, 2025-03-04:
Hi @xuankuzcr. Following the code, the NCC seems to be disabled.

https://github.com/hku-mars/FAST-LIVO2/blob/162f843fa5c6a2903baa0d2d9bf16c1b287834c9/include/vio.h#L101

Although I enabled it and set `ncc_thre=0.9`, I did not observe noticeable differences in the experimental results visually. Do you have any tuning experience for the `ncc_thre`?

xuankuzcr, 2025-03-05:
Generally, the smaller the NCC value, the more points are filtered out. You could try testing with scenes that have more depth-discontinuous edges, using thresholds of 0.3 and 0.4—the difference might be more noticeable (By the way, I haven't even enabled it yet, and you're already delving into research).

DapengFeng, 2025-03-06:
Thanks for the insight! The lower NCC thresholds (like 0.3/0.4) will indeed prune more points, especially in scenes with sharp depth jumps where mismatches are likelier.  (And haha, guilty as charged! Your proactive deep-dive has me excited to see results once the feature’s enabled. Keep the ideas coming! 🚀😄)