hku-mars/FAST-LIVO2 #136: LO模式

gaoyi-2016, 2025-03-04:
您好，郑博，LO模式似乎仍然存在一些问题

[image]

#100

xuankuzcr, 2025-03-05:
在`LO`模式下，通常需要提供去畸变后的点云，或者要求在采集数据时保持缓慢运动。`red_sculpture`这个数据包运动相对剧烈，你可以用 [这个数据](https://connecthkuhk-my.sharepoint.com/:f:/g/personal/zhengcr_connect_hku_hk/EqXewSlNCl5OnNxFMo3aCCQBpLQ4jjQaQ9IRCvq0dmPueg?e=Pyyalf) 做测试。此外，我添加了匀速运动模型去畸变，这样效果会更好一些。
```yaml
common:
  img_topic: "/left_camera/image"
  lid_topic: "/livox/lidar"
  imu_topic: "/livox/imu"
  img_en: 0
  lidar_en: 1
  ros_driver_bug_fix: false

extrin_calib:
  extrinsic_T: [0.04165, 0.02326, -0.0284]
  extrinsic_R: [1, 0, 0, 0, 1, 0, 0, 0, 1]
  Rcl: [0.00610193,-0.999863,-0.0154172,
        -0.00615449,0.0153796,-0.999863,
        0.999962,0.00619598,-0.0060598] 
  Pcl: [0.0194384, 0.104689,-0.0251952]

time_offset: 
  imu_time_offset: 0.0
  img_time_offset: 0.1
  exposure_time_init: 0.0

preprocess:
  point_filter_num: 1
  filter_size_surf: 0.1
  lidar_type: 1 # Livox Avia LiDAR
  scan_line: 6
  blind: 0.8

vio:
  max_iterations: 5
  outlier_threshold: 1000 # 78 100 156 #100 200 500 700 infinite
  img_point_cov: 100 # 100 1000 
  patch_size: 8
  patch_pyrimid_level: 4
  normal_en: true
  raycast_en: false
  inverse_composition_en: false
  exposure_estimate_en: true
  inv_expo_cov: 0.1

imu:
  imu_en: false
  imu_int_frame: 30
  acc_cov: 0.5 # 0.2
  gyr_cov: 10.0 # 0.5
  b_acc_cov: 0.0001 # 0.1
  b_gyr_cov: 0.0001 # 0.1

lio:
  max_iterations: 5
  dept_err: 0.02
  beam_err: 0.05
  min_eigen_value: 0.01 # 0.005
  voxel_size: 0.5
  max_layer: 2
  max_points_num: 50
  layer_init_num: [5, 5, 5, 5, 5]

local_map:  
  map_sliding_en: false
  half_map_size: 100
  sliding_thresh: 8

uav:
  imu_rate_odom: false
  gravity_align_en: false

publish:
  dense_map_en: true
  pub_effect_point_en: false 
  pub_plane_en: false
  pub_scan_num: 1
  blind_rgb_points: 0.0

evo:
  seq_name: "CBD_Building_01"
  pose_output_en: false

pcd_save:
  pcd_save_en: false
  colmap_output_en: false # need to set interval = -1
  filter_size_pcd: 0.15
  interval: -1
  # how many LiDAR frames saved in each pcd file;
  # -1 : all frames will be saved in ONE pcd file, may lead to memory crash when having too much frames.
```

gaoyi-2016, 2025-03-05:
已经测试, 该数据，LO模式可以正常运行