hku-mars/FAST-LIVO2 #140: 在激光点云帧和图像帧同步的情况下，图像帧到达的时间为什么需要再加一个偏移

nowPB, 2025-03-05:
在LIVMapper.cpp文件中图像接收回调函数img_cbk中，图像帧的到达时间为 
 double msg_header_time = msg->header.stamp.toSec() + img_time_offset;
其中yaml文件显示img_time_offset=0.1。为什么需要加一个0.1秒的偏移量？

prethn, 2025-03-05:
img_time_offset 这个偏移量主要是用来调整不同步的情况，在激光点云帧和图像帧同步情况下设为0。

G4419, 2025-03-12:
请问下怎么测量激光和图像的offset呢

JixinGao, 2025-04-03:
@xuankuzcr 郑博能麻烦您看下这个issue嘛？不太理解img_time_offset: 0.1和 lidar_time_offset: -0.1，因为看到您最新适配的ntu数据集也是增加了一个lidar的time offset，为什么需要一个时间偏移且偏移为什么是0.1/-0.1呢？我去掉这个时间偏移，在ntu数据集上进行测试，精度就有一些损失。另外想问一下您，轨迹评估时好像时间上并没完全对上，这是没关系的嘛？
以下直接用您给的轨迹做的evo_ape ... -a -p
[image]

xuankuzcr, 2025-04-03:
这个问题新开个issue吧，估计有这个问题的人挺多的@JixinGao

JixinGao, 2025-04-03:
好的好的

yccckid, 2026-08-24:
> 这个新开个issue吧，估计有这个问题的人挺多的[@JixinGao](https://github.com/JixinGao)

请问有链接吗

JixinGao, 2026-08-25:
> > 这个新开个issue吧，估计有这个问题的人挺多的[@JixinGao](https://github.com/JixinGao)
> 
> 请问有链接吗

#202