hku-mars/FAST-LIVO2 #147: [laserMapping-2] process has died

1391741823, 2025-03-06:
出现了编译错误 有没有老哥碰到了跟我一样的呢

[laserMapping-2] process has died [pid 22063, exit code -6, cmd /home/ao/jiguang/LIVO2/ws_livo2/devel/lib/fast_livo/fastlivo_mapping __name:=laserMapping __log:=/home/ao/.ros/log/87da1cb4-fa95-11ef-83e9-77506fa045ba/laserMapping-2.log].
log file: /home/ao/.ros/log/87da1cb4-fa95-11ef-83e9-77506fa045ba/laserMapping-2*.log

[image]

图像有第一帧图像 vio与lio都有原始点云存在

prexhu, 2025-03-07:
你可以看看这个hku-mars/FAST-LIVO#173, 希望能帮到你

1391741823, 2025-03-07:
> 建议可以在mapping的launch文件中加入launch-prefix,看看gdb信息 如果不奏效的话可以参考这个[hku-mars/FAST-LIVO#173](https://github.com/hku-mars/FAST-LIVO/issues/173), 希望能帮到你

你好我按照方法加入了launch-prefix,依旧报错为超出字符串长度 晚点我试试其他方法 感谢指导

[image]

prexhu, 2025-03-07:
> > 建议可以在mapping的launch文件中加入launch-prefix,看看gdb信息 如果不奏效的话可以参考这个[hku-mars/FAST-LIVO#173](https://github.com/hku-mars/FAST-LIVO/issues/173), 希望能帮到你
> 
> 你好我按照方法加入了launch-prefix,依旧报错为超出字符串长度 晚点我试试其他方法 感谢指导
> 
> [image]

你这个报错和我当时使用launch-prefix的报错很相似,都是很底层的问题,我是通过重装pcl库解决的,并且多次出现了类似错误,都通过重装pcl库解决了,详细信息参考我提到的issue,祝你成功复现😎

1391741823, 2025-03-09:
> 你这个报错和我当时使用launch-prefix的报错很相似,都是很底层的问题,我是通过重装pcl库解决的,并且多次出现了类似错误,都通过重装pcl库解决了,详细信息参考我提到的issue,祝你成功复现😎

谢谢提醒 我重装了pcl库依旧没有改变 也重装了opencv4.5.3 今天突然发现我每次启动roslaunch的时候会报错下面的东西：

[image]

内存少于1gb不知道是不是这个原因 我再调试试试

prexhu, 2025-03-09:
> 谢谢提醒 我重装了pcl库依旧没有改变 也重装了opencv4.5.3 今天突然发现我每次启动roslaunch的时候会报错下面的东西：
>
> [image]
>
> 内存少于1gb不知道是不是这个原因 我再调试试试

opencv 的版本应该不用那么高
内存少于1gb是ros已经存储的的日志文件小于1gb

1391741823, 2025-03-10:
> opencv 的版本应该不用那么高 内存少于1gb是ros已经存储的的日志文件小于1gb

谢谢老哥我换了一台电脑把环境重新装一遍 可以运行了 不是这个log文件的过小的原因 还是会报过小哈哈 感谢哥们答疑解惑 祝科研顺利！

[image]

1315hly, 2025-03-14:
你好，我在复现这个工作时也遇到以下问题：[laserMapping-2] process has died [pid 5921, exit code -11, cmd /home/xiao/FAST-LIVO2/catkin_ws/devel/lib/fast_livo/fastlivo_mapping __name:=laserMapping __log:=/home/xiao/.ros/log/2d7e2436-009e-11f0-8c7f-291444ddf871/laserMapping-2.log].
log file: /home/xiao/.ros/log/2d7e2436-009e-11f0-8c7f-291444ddf871/laserMapping-2*.log
不过我使用GDB调试后的问题如下：Starting program: /home/xiao/FAST-LIVO2/catkin_ws/devel/lib/fast_livo/fastlivo_mapping
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffe254a700 (LWP 5325)]
[New Thread 0x7fffe1d49700 (LWP 5326)]
[New Thread 0x7fffe1548700 (LWP 5327)]
[New Thread 0x7fffe0d47700 (LWP 5328)]

Thread 1 "fastlivo_mappin" received signal SIGSEGV, Segmentation fault.
__GI___libc_free (mem=0x1) at malloc.c:3102
3102 malloc.c: 没有那个文件或目录.
(gdb) bt
#0 __GI___libc_free (mem=0x1) at malloc.c:3102
https://github.com/hku-mars/FAST-LIVO2/issues/1 0x00007ffff5ae62df in std::__cxx11::basic_string<char, std::char_traits, std::allocator >::_M_assign(std::__cxx11::basic_string<char, std::char_traits, std::allocator > const&) ()
at /lib/x86_64-linux-gnu/libstdc++.so.6
https://github.com/hku-mars/FAST-LIVO2/issues/2 0x00007ffff5d28bda in boost::filesystem::detail::directory_iterator_construct(boost::filesystem::directory_iterator&, boost::filesystem::path const&, boost::system::error_code*) () at /usr/local/lib/libboost_filesystem.so.1.65.0
https://github.com/hku-mars/FAST-LIVO2/issues/3 0x00007fffea5f79b0 in rospack::Rosstackage::isStackage(std::__cxx11::basic_string<char, std::char_traits, std::allocator > const&) ()
at /opt/ros/noetic/lib/librospack.so
https://github.com/hku-mars/FAST-LIVO2/issues/4 0x00007fffea5fea3f in rospack::Rosstackage::crawlDetail(std::__cxx11::basic_string<char, std::char_traits, std::allocator > const&, bool, int, bool, std::vector<rospack::DirectoryCrawlRecord*, std::allocatorrospack::DirectoryCrawlRecord* >&, boost::unordered::unordered_set<std::__cxx11::basic_string<char, std::char_traits, std::allocator >, boost::hash<std::__cxx11::basic_string<char, std::char_traits, std::allocator > >, std::equal_to<std::__cxx11::basic_string<char, std::char_traits, std::allocator > >, std::allocator<std::__cxx11::basic_string<char, std::char_traits, std::allocator > > >&) () at /opt/ros/noetic/lib/librospack.so
https://github.com/hku-mars/FAST-LIVO2/issues/5 0x00007fffea6006be in rospack::Rosstackage::crawl(std::vector<std::__cxx11::basic_string<char, std::char_traits, std::allocator >, std::allocator<std::__cxx11::basic_string<char, std::char_traits, std::allocator --Type for more, q to quit, c to continue without paging--
所以想请教一下这个问题该如何解决

1391741823, 2025-03-14:
> 你好，我在复现这个工作时也遇到以下问题：[laserMapping-2] process has died [pid 5921, exit code -11, cmd /home/xiao/FAST-LIVO2/catkin_ws/devel/lib/fast_livo/fastlivo_mapping __name:=laserMapping [...]

https://github.com/hku-mars/FAST-LIVO/issues/143#issue-2634801075 
亲看看这个有没有帮助

1315hly, 2025-03-14:
> > 你好，我在复现这个工作时也遇到以下问题：[laserMapping-2] 进程已死 [pid 5921， exit code -11， cmd /home/xiao/FAST-LIVO2/catkin_ws/devel/lib/fast_livo/fastlivo_mapping __name：=laserMapping __log：=/home/xiao/.ros/log/2d7e2436-009e-11f0-8c7f-291444ddf871/laserMapping-2.log]。日志文件： /home/xiao/.ros/log/2d7e2436-009e-11f0-8c7f-291444ddf871/laserMapping-2*.log 不过我使用GDB调试后的问题如下：启动程序： /home/xiao/FAST-LIVO2/catkin_ws/devel/lib/fast_livo/fastlivo_mapping [启用使用libthread_db进行线程调试] 使用主机libthread_db库 “/lib/x86_64-linux-gnu/libthread_db.so.1”。[新线程0x7fffe254a700 （LWP 5325）][新线程0x7fffe1d49700 （LWP 5326）][新线程0x7fffe1548700 （LWP 5327）][新线程0x7fffe0d47700 （LWP 5328）]
> > 线程 1 “fastlivo_mappin” 接收到信号 SIGSEGV，分段错误。__GI___libc_free （mem=0x1） at malloc.c：3102 3102 malloc.c： 没有那个文件或目录。（gdb） bt #0 __GI___libc_free （mem=0x1） at malloc.c：3102 [#1](https://github.com/hku-mars/FAST-LIVO2/issues/1) 0x00007ffff5ae62df in std：：__cxx11：：basic_string<char， std：：char_traits， std：：allocator >：：_M_assign（std：：__cxx11：：：：c basic_string<char， std：：char_traits， std：：allocator > const&） （） 在 /lib/x86_64-linux-gnu/libstdc++.so.6 [#2](https://github.com/hku-mars/FAST-LIVO2/issues/2) 0x00007ffff5d28bda in boost：：filesystem：:d etail：:d irectory_iterator_construct（boost：：filesystem：:d irectory_iterator&， boost：：filesystem：:p ath const&， boost：：system：：error_code*） （） 在 /usr/local/lib/libboost_filesystem.so.1.65.0 [#3](https://github.com/hku-mars/FAST-LIVO2/issues/3) 0x00007fffea5f79b0在 rospack：：Rosstackage：：isStackage（std：：__cxx11：：isStackage（std：：basic_string<：：char， std：：char_traits， std：：allocator > const&） （） 在 /opt/ros/noetic/lib/librospack.so [#4](https://github.com/hku-mars/FAST-LIVO2/issues/4)0x00007fffea5fea3f rospack：：Rosstackage：：crawlDetail（std：：__cxx11：：basic_string<char， std：：char_traits， std：：allocator > const&， bool， int， bool， std：：vector<rospack：:D irectoryCrawlRecord*， std：：allocatorrospack：:D irectoryCrawlRecord* >&， boost：：unordered：：unordered_set<std：：__cxx11：：basic_string<char， std：：char_traits， std：：allocator >， boost：：hash<std：：__cxx11：：basic_string<char， std：：char_traits， std：：allocator > >， std：：equal_to<std：：__cxx11：：basic_string<char， std：：char_traits， std：：allocator > >， std：：allocator<std：：__cxx11：：basic_string<char， std：：char_traits， std：：allocator > > >&） （） 在 /opt/ros/noetic/lib/librospack.so [#5](https://github.com/hku-mars/FAST-LIVO2/issues/5) 0x00007fffea6006be in rospack：：Rosstackage：：crawl（std：：vector<std：：__cxx11：：basic_string<char， std：：char_traits， std：：allocator >， std：：allocator<std：：__cxx11：：basic_string<char， std：：char_traits， std：：allocator --键入更多， q 退出， c 继续而不分页-- 所以想请教一下这个问题该如何解决
> 
> [hku-mars/FAST-LIVO#143 （comment）](https://github.com/hku-mars/FAST-LIVO/issues/143#issue-2634801075) 亲看看这个有没有帮助

非常感谢你的回答，不过我并没有在fast-livo2中找到laserMapping.cpp，且在仅有的文件中没找到initUndistortRectifyMap相关的信息