hku-mars/FAST-LIVO2 #148: 关于分帧及sequential update相关问题请教

JixinGao, 2025-03-07:
郑博您好，首先感谢开源这么惊艳的工作！
有几个点想请教一下，
关于论文：
1）point-lio分帧以点频更新可以从根本上免除畸变补偿，我感觉在剧烈运动下还是很能提高效果的。fast-livo2也进行了分帧，但为什么没有顺便考虑逐点update呢，想听一下您的看法。
2）看别的issue里您说fast-livo2分帧是为了相机和LiDAR帧对齐，可不是已经做了硬件触发的帧对齐了嘛，是为了对齐的更准确嘛？
此外我看了ntu的数据集，图像和雷达是没有时间戳对齐的，分帧是针对这类情形嘛？
3）另外想请教一下，看了您supplementary中测试的三种更新方式后没太理解，我理解是这样的：
第一种“asynchronous+standard update”是时间戳没有对齐，来了哪个观测就做哪个观测（类似imu预测+lidar观测+imu预测+camera观测）；
第二种“sync+standard update”是对齐了时间戳后，imu预测+((lidar+camera)共同的update)；
第三种 sequential update更新是时间戳对齐后，imu预测+lidar更新再做camera更新。
不知道理解的是否有误，麻烦了！
4）还有一个小问题想向您确定一下，如果使用360°LiDAR，是只有和图像重合的视角内的点云被用到了嘛？
再次感谢您的工作，期待帮助！🤩

xuankuzcr, 2025-03-07:
您好，我逐一回复下
1）fast-livo2分帧是为了把相机和LiDAR帧对齐，然后去做sequential update，而point-lio以点频去更新是为了抵抗IMU饱和/超量程的情况，二者motivation不一样。如果单讨论对逐点update的看法，我认为这种更新方式易受LiDAR噪点的影响，估计出来的速度抖动会很大，因此在没那么剧烈的运动下 (小于20m/s2) 效果不一定好。而按帧去估计位姿，就没那么容易受outlier的影响。除此之外，正常的机器人应用里超过10m/s2的运动都很少见。综上，最开始的系统设计中没有考虑这种update方式。
2）硬同步的数据不一定是要时间戳完全对齐的，只要各个sensor的采样时间戳在同一个时间轴上就可以。因此分帧确实是为了针对相机和LiDAR不同频率以及二者时间戳没对齐的数据的。
3）正确。
4）不是的，visual module是只用图像FoV内的地图点，而LiDAR module是360°的都用了，只不过rviz中可视化只能看到图像FoV内的点云。

JixinGao, 2025-03-07:
非常非常感谢您的回复！
关于数据时间戳对齐这里，还有一点疑惑：
您已经设计了分帧来处理传感器频率不同（同一时间轴，时间戳没对齐）的情况，我看您还是比较推荐硬件触发的时间戳对齐，private的数据也都是时间戳对齐。
那么对于FAST-LIVO2来说，是不是 **硬件触发时间戳对齐的情况** 是要优于 **没对齐的情况**？（毕竟硬件触发的更准确嘛，即使已经有了分帧啦）
最后，可以这样理解嘛——ntu数据集中那种没对齐时间戳的数据，也算是能把FAST-LIVO2跑的很好，只是如果硬触发时间戳对齐数据后，可以更好。
感谢您的解答，麻烦啦！

xuankuzcr, 2025-03-07:
只要时间戳是传感器真实采样的时间戳，是否对齐不关键，也不影响最终效果，fast-livo2里面会自动分帧来对齐数据。

JixinGao, 2025-03-07:
好的明白了！感谢！

closed 2025-03-07 by JixinGao