hku-mars/FAST-LIVO2 #151: 点云与RGB结合失败

autumntang, 2025-03-08:
郑博您好！我运行fast_livo2得到的结果存在一些问题，具体表现如下：

[image]

[image]
我使用的传感器为mid360以及d435i，已经做过外参标定和时间同步，以下是时间同步后的lidar与相机timestamp效果：

[image]

[image]
想向您请教，出现以上结果，可能是在什么地方出了问题。
并且，以下是我使用的mid360.yaml参数：common:
  img_topic: "/color"
  lid_topic: "/livox/lidar"
  imu_topic: "/livox/imu"
  img_en: 1
  lidar_en: 1
  ros_driver_bug_fix: false

extrin_calib:
  extrinsic_T: [-0.011, -0.02329, 0.04412]          #
  extrinsic_R: [1, 0, 0, 0, 1, 0, 0, 0, 1]
  Rcl: [-0.0385484, 0.0582421, 0.9975579,
 -0.9992565, -0.0032215, -0.0384259,
 0.0009756, -0.9982976, 0.0583231] 
  Pcl: [0.0191877, 0.0314342, -0.0181279]

time_offset:
  imu_time_offset: 0.0
  img_time_offset: 0.1
  exposure_time_init: 0.0

preprocess:
  point_filter_num: 1
  filter_size_surf: 0.1
  lidar_type: 1 # Livox Avia LiDAR
  scan_line: 4
  blind: 0.5

vio:
  max_iterations: 5
  outlier_threshold: 1000 # 78 100 156 #100 200 500 700 infinite
  img_point_cov: 100 # 100 1000
  patch_size: 8
  patch_pyrimid_level: 4
  normal_en: true
  raycast_en: false
  inverse_composition_en: false
  exposure_estimate_en: true
  inv_expo_cov: 0.1

imu:
  imu_en: true
  imu_int_frame: 30
  acc_cov: 0.8 # 0.2
  gyr_cov: 0.5 # 0.5
  b_acc_cov: 0.001 # 0.1
  b_gyr_cov: 0.001 # 0.1

lio:
  max_iterations: 5
  dept_err: 0.02
  beam_err: 0.15
  min_eigen_value: 0.005 # 0.005
  voxel_size: 0.5
  max_layer: 2
  max_points_num: 50
  layer_init_num: [5, 5, 5, 5, 5]

local_map:
  map_sliding_en: false             #false
  half_map_size: 100
  sliding_thresh: 8

uav:
  imu_rate_odom: false
  gravity_align_en: true

publish:
  dense_map_en: true
  pub_effect_point_en: false
  pub_plane_en: false
  pub_scan_num: 1
  blind_rgb_points: 0.0

evo:
  seq_name: "CBD_Building_01"
  pose_output_en: false

pcd_save:
  pcd_save_en: true
  colmap_output_en: true # need to set interval = -1
  filter_size_pcd: 0.15
  interval: -1
文件中的extrinsic_T和extrinsic_R我是根据您在另一个issue中的回答所设置，以及还想请问您，该参数是为什么设置为上面的值，与Rcl、Pcl有何不同，Rcl与Pcl是相机到雷达的旋转矩阵和平移矩阵，extrinsic_T和extrinsic_R代表了什么？
对于以上诸多问题我深感抱歉，诚盼您的回复！

autumntang, 2025-03-09:
补充：采集的第一帧rgb与lidar融合效果比较好，移动lidar和相机之后rviz中显示的点云上附着的rgb肉眼可见的变模糊

gc625, 2025-03-09:
如果第一帧rgb与lidar融合效果比较好，那我觉得应该是cam和lidar同步没做好，rqt_bag里要看lidar msg 和 cam msg header里的timestamp，而不只是看msg在時間軸中的位置

autumntang, 2025-03-09:
感谢您的回答！我选择了如下两个时间戳，这二者之间的时间差距是否过大？此外我发现，尽管我设置相机的频率也为10hz，但从时间戳上看来相机与lidar之间每次发布消息的时间戳差值并不是固定的，请问您是否知道这是什么原因
[image]

tanble, 2025-03-10:
你可以将相机往后偏移一帧看一下，如果不行再验证你的相机是否是10hz

autumntang, 2025-03-10:
感谢您的回复！数据已发送到您邮箱

xuankuzcr, 2025-03-11:
我测试了你的数据，静止时的效果就很奇怪，看起来外参有问题，能否提供下您在静止时上色的截图

autumntang, 2025-03-12:
您好，下面是静止时的截图：
[image]

[image]

[image]

xuankuzcr, 2025-03-17:
[image]
抱歉回复晚了，看起来内外参有问题，前面那个女生椅子靠背上的棕色外套都上色到墙上 了

autumntang, 2025-03-17:
@xuankuzcr 好的谢谢，那么时间同步呢？跟时间同步没有关系吗

ahui66686, 2025-05-17:
> 郑博您好！我运行fast_livo2得到的结果存在一些问题，具体表现如下：
>
> [image]
>
> [image] 我使用的传感器为mid360以及d435i，已经做过外参标定和时间同步，以下是时间同步后的lidar与相机timestamp效果：
>
> [image]
>
> [image] 想向您请教，出现以上结果，可能是在什么地方出了问题。 并且，以下是我使用的mid360.yaml参数：common: [...]

> 郑博您好！我运行fast_livo2得到的结果存在一些问题，具体表现如下：
> 
> [image]
> 
> [image] 我使用的传感器为mid360以及d435i，已经做过外参标定和时间同步，以下是时间同步后的lidar与相机timestamp效果：
> 
> [image]
> 
> [image] 想向您请教，出现以上结果，可能是在什么地方出了问题。 并且，以下是我使用的mid360.yaml参数：common: img_topic: "/color"  img_topic：“/color” lid_topic: "/livox/lidar"lid_topic：“/livox/lidar” imu_topic: "/livox/imu"  imu_topic：“/livox/IMU” img_en: 1  img_en：1 lidar_en: 1  lidar_en：1 ros_driver_bug_fix: falseros_driver_bug_fix： false
> 
> extrin_calib:  extrin_calib： extrinsic_T: [-0.011, -0.02329, 0.04412] #extrinsic_T： [-0.011， -0.02329， 0.04412]# extrinsic_R: [1, 0, 0, 0, 1, 0, 0, 0, 1]extrinsic_R： [1， 0， 0， 0， 1， 0， 0， 0， 1] Rcl: [-0.0385484, 0.0582421, 0.9975579,Rcl： [-0.0385484， 0.0582421， 0.9975579， -0.9992565, -0.0032215, -0.0384259, 0.0009756, -0.9982976, 0.0583231] Pcl: [0.0191877, 0.0314342, -0.0181279]Pcl： [0.0191877， 0.0314342， -0.0181279]
> 
> time_offset:  time_offset： imu_time_offset: 0.0  imu_time_offset：0.0 img_time_offset: 0.1  img_time_offset：0.1 exposure_time_init: 0.0  exposure_time_init：0.0
> 
> preprocess:  预处理： point_filter_num: 1  point_filter_num：1 filter_size_surf: 0.1  filter_size_surf：0.1 lidar_type: 1 # Livox Avia LiDARlidar_type：1 # Livox Avia 激光雷达 scan_line: 4  scan_line：4 blind: 0.5  盲注： 0.5
> 
> vio:  锯： max_iterations: 5  max_iterations：5 outlier_threshold: 1000 # 78 100 156 [#100](https://github.com/hku-mars/FAST-LIVO2/issues/100) 200 500 700 infiniteoutlier_threshold：1000 # 78 100 156 [#100](https://github.com/hku-mars/FAST-LIVO2/issues/100) 200 500 700 无限 img_point_cov: 100 # 100 1000img_point_cov：100 # 100 1000 patch_size: 8  patch_size：8 patch_pyrimid_level: 4  patch_pyrimid_level：4 normal_en: true  normal_en：true raycast_en: false  raycast_en：false inverse_composition_en: falseinverse_composition_en：false exposure_estimate_en: trueexposure_estimate_en：true inv_expo_cov: 0.1  inv_expo_cov：0.1
> 
> imu:  IMU： imu_en: true  imu_en：true imu_int_frame: 30  imu_int_frame：30 acc_cov: 0.8 # 0.2acc_cov：0.8 # 0.2 gyr_cov: 0.5 # 0.5gyr_cov：0.5 # 0.5 b_acc_cov: 0.001 # 0.1b_acc_cov：0.001 # 0.1 b_gyr_cov: 0.001 # 0.1b_gyr_cov：0.001 # 0.1
> 
> lio:  廖： max_iterations: 5  max_iterations：5 dept_err: 0.02  dept_err：0.02 beam_err: 0.15  beam_err：0.15 min_eigen_value: 0.005 # 0.005min_eigen_value：0.005 # 0.005 voxel_size: 0.5  voxel_size：0.5 max_layer: 2  max_layer：2 max_points_num: 50  max_points_num：50 layer_init_num: [5, 5, 5, 5, 5]layer_init_num：[5， 5， 5， 5， 5]
> 
> local_map:  local_map： map_sliding_en: false #falsemap_sliding_en：false #false half_map_size: 100  half_map_size：100 sliding_thresh: 8  sliding_thresh：8
> 
> uav:  无人机： imu_rate_odom: false  imu_rate_odom：false gravity_align_en: true  gravity_align_en：true
> 
> publish:  发布： dense_map_en: true  dense_map_en：true pub_effect_point_en: falsepub_effect_point_en： false pub_plane_en: false  pub_plane_en：false pub_scan_num: 1  pub_scan_num：1 blind_rgb_points: 0.0  blind_rgb_points：0.0
> 
> evo:  这里： seq_name: "CBD_Building_01"seq_name： “CBD_Building_01” pose_output_en: false  pose_output_en：false
> 
> pcd_save:  pcd_save： pcd_save_en: true  pcd_save_en：true colmap_output_en: true # need to set interval = -1colmap_output_en： true # 需要设置 interval = -1 filter_size_pcd: 0.15  filter_size_pcd：0.15 interval: -1  间隔：-1 文件中的extrinsic_T和extrinsic_R我是根据您在另一个issue中的回答所设置，以及还想请问您，该参数是为什么设置为上面的值，与Rcl、Pcl有何不同，Rcl与Pcl是相机到雷达的旋转矩阵和平移矩阵，extrinsic_T和extrinsic_R代表了什么？ 对于以上诸多问题我深感抱歉，诚盼您的回复！

你好，我想请教一下时间同步的问题，我的vx是13099004163，希望得到您的回复