hku-mars/FAST-LIVO2 #152: Fast-livo2 localisation only?

WheresWes, 2025-03-09:
Fast- lio had a localisation only version that allowed you to localise within a pre built map.  Can this be done with fast-livo2?

Alexanderofhust, 2025-03-10:
The same demand, sincerely look forward to Dr. Zheng  answer

xuankuzcr, 2025-03-11:
Currently, we do not plan to implement the pure localization mode in the immediate future. However, this remains part of our long-term development goals. In the short term, we are actively working on an LIV-init (LiDAR-Inertial-Visual initialization) module that will be added to this repository.

WheresWes, 2025-03-11:
Thankyou for the update.. I'm glad to hear that the localisation is planned even if it is a long term goal.  I'm interested in this initialization module.. what is the intended function?

xuankuzcr, 2025-03-11:
Initialize the extrinsic and time offset between the camera and LiDAR, LiDAR and IMU, as well as the IMU's bias and noise..

WheresWes, 2025-03-11:
Ahh yes I can see that being helpful!  Quick question while I have you captive.  Is it mandatory for the camera to have global shutter? Or is a high refresh rate rolling shutter with hardware sync adequate?

xuankuzcr, 2025-03-11:
Currently, a global shutter is mandatory. Support for rolling shutter is also under development.

WheresWes, 2025-03-11:
You lads are busy! Is there a public road map available?

JerryXiao1314, 2026-02-26:
> Fast- lio had a localisation only version that allowed you to localise within a pre built map. Can this be done with fast-livo2?

Welcome to follow our work [https://github.com/HorizonRobotics/HoloAgent](url). In this framework, we use modified fastlivo2 to build map and localise within the pre-built map.