hku-mars/FAST-LIVO2 #154: compile

BruceYin888, 2025-03-11:
/usr/bin/ld: /home/robot/fast_livo2/devel/lib/libvikit_common.so: undefined reference to `Sophus::SE3::exp(Eigen::Matrix<double, 6, 1, 0, 6, 1> const&)'
/usr/bin/ld: /home/robot/fast_livo2/devel/lib/libvikit_common.so: undefined reference to `Sophus::SE3::operator=(Sophus::SE3 const&)'
/usr/bin/ld: /home/robot/fast_livo2/devel/lib/libvikit_common.so: undefined reference to `Sophus::SE3::SE3()'
/usr/bin/ld: /home/robot/fast_livo2/devel/lib/libvikit_common.so: undefined reference to `Sophus::SE3::SE3(Sophus::SE3 const&)'
/usr/bin/ld: /home/robot/fast_livo2/devel/lib/libvikit_common.so: undefined reference to `Sophus::SE3::operator*(Sophus::SE3 const&) const'
/usr/bin/ld: /home/robot/fast_livo2/devel/lib/libvikit_common.so: undefined reference to `Sophus::SO3::matrix() const'
/usr/bin/ld: /home/robot/fast_livo2/devel/lib/libvikit_common.so: undefined reference to `Sophus::SE3::SE3(Eigen::Matrix<double, 3, 3, 0, 3, 3> const&, Eigen::Matrix<double, 3, 1, 0, 3, 1> const&)'
/usr/bin/ld: /home/robot/fast_livo2/devel/lib/libvikit_common.so: undefined reference to `Sophus::SE3::operator*(Eigen::Matrix<double, 3, 1, 0, 3, 1> const&) const'
collect2: error: ld returned 1 exit status
/usr/bin/ld: /home/robot/fast_livo2/devel/lib/libvikit_common.so: undefined reference to `Sophus::SE3::exp(Eigen::Matrix<double, 6, 1, 0, 6, 1> const&)'
/usr/bin/ld: /home/robot/fast_livo2/devel/lib/libvikit_common.so: undefined reference to `Sophus::SE3::operator=(Sophus::SE3 const&)'
/usr/bin/ld: /home/robot/fast_livo2/devel/lib/libvikit_common.so: undefined reference to `Sophus::SE3::SE3()'
/usr/bin/ld: /home/robot/fast_livo2/devel/lib/libvikit_common.so: undefined reference to `Sophus::SE3::SE3(Sophus::SE3 const&)'
/usr/bin/ld: /home/robot/fast_livo2/devel/lib/libvikit_common.so: undefined reference to `Sophus::SE3::operator*(Sophus::SE3 const&) const'
/usr/bin/ld: /home/robot/fast_livo2/devel/lib/libvikit_common.so: undefined reference to `Sophus::SO3::matrix() const'
/usr/bin/ld: /home/robot/fast_livo2/devel/lib/libvikit_common.so: undefined reference to `Sophus::SE3::SE3(Eigen::Matrix<double, 3, 3, 0, 3, 3> const&, Eigen::Matrix<double, 3, 1, 0, 3, 1> const&)'
/usr/bin/ld: /home/robot/fast_livo2/devel/lib/libvikit_common.so: undefined reference to `Sophus::SE3::operator*(Eigen::Matrix<double, 3, 1, 0, 3, 1> const&) const'
make[2]: *** [rpg_vikit/vikit_common/CMakeFiles/test_vk_common_camera.dir/build.make:134: /home/robot/fast_livo2/devel/lib/vikit_common/test_vk_common_camera] Error 1
collect2: error: ld returned 1 exit status
make[1]: *** [CMakeFiles/Makefile2:575: rpg_vikit/vikit_common/CMakeFiles/test_vk_common_camera.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: *** [rpg_vikit/vikit_common/CMakeFiles/test_vk_common_triangulation.dir/build.make:134: /home/robot/fast_livo2/devel/lib/vikit_common/test_vk_common_triangulation] Error 1
make[1]: *** [CMakeFiles/Makefile2:1007: rpg_vikit/vikit_common/CMakeFiles/test_vk_common_triangulation.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
Invoking "make -j16 -l16" failed
I want to ask why this happened? how can I solve the problems

xuankuzcr, 2025-03-11:
Please install the dependencies according to the README.

BruceYin888, 2025-03-11:
> Please install the dependencies according to the README.

I have already installed the dependencies, 我按照官网的教程安装的sophus

BruceYin888, 2025-03-11:
> Please install the dependencies according to the README.

你好，郑博，依赖我都按照要求安装了这个报错好像显示sophus有问题

Qiushui3, 2025-03-12:
好像是需要在fastlivo2的Cmakelists里面加一句set(Sophus LIBRARIES "/usr/local/lib/libSophus.so")，如果不行的话我记得还需要在vikit的cmakelist中也加一句

BruceYin888, 2025-03-12:
> 话我记得还需要在vikit的cmakelist中也加一句

fastlivo2我加了，vikit的没加，不知道加在什么地方

Qiushui3, 2025-03-12:
> > 话我记得还需要在vikit的cmakelist中也加一句
> 
> fastlivo2我加了，vikit的没加，不知道加在什么地方

我貌似复制错了，就在最后一个回复中，我通过这样的操作解决了问题

BruceYin888, 2025-03-12:
> > > 话我记得还需要在vikit的cmakelist中也加一句
> > 
> > 
> > fastlivo2我加了，vikit的没加，不知道加在什么地方
> 
> 我貌似复制错了，就在最后一个回复中，我通过这样的操作解决了问题

编译成了，但是运行有些问题

Qiushui3, 2025-03-12:
> > > > 话我记得还需要在vikit的cmakelist中也加一句
> > > 
> > > 
> > > fastlivo2我加了，vikit的没加，不知道加在什么地方
> > 
> > 
> > 我貌似复制错了，就在最后一个回复中，我通过这样的操作解决了问题
> 
> 编译成了，但是运行有些问题

你可以再试一下安装compressed_image_transport插件 “sudo apt-get install ros-noetic-compressed-image-transport”

BruceYin888, 2025-03-12:
> > > > > 话我记得还需要在vikit的cmakelist中也加一句
> > > > 
> > > > 
> > > > fastlivo2我加了，vikit的没加，不知道加在什么地方
> > > 
> > > 
> > > 我貌似复制错了，就在最后一个回复中，我通过这样的操作解决了问题
> > 
> > 
> > 编译成了，但是运行有些问题
> 
> 你可以再试一下安装compressed_image_transport插件 “sudo apt-get install ros-noetic-compressed-image-transport”

robot@LAPTOP-18QMGK0H:~$ sudo apt-get install ros-noetic-compressed-image-transport
[sudo] password for robot:
Reading package lists... Done
Building dependency tree
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 ros-noetic-compressed-image-transport : Depends: libturbojpeg0-dev but it is not going to be installed
E: Unable to correct problems, you have held broken packages.安不上报错，老哥方便加个联系方式摄入交流一下吗？

Qiushui3, 2025-03-12:
> > > > > > 话我记得还需要在vikit的cmakelist中也加一句
> > > > > 
> > > > > 
> > > > > fastlivo2我加了，vikit的没加，不知道加在什么地方
> > > > 
> > > > 
> > > > 我貌似复制错了，就在最后一个回复中，我通过这样的操作解决了问题
> > > 
> > > 
> > > 编译成了，但是运行有些问题
> > 
> > 
> > 你可以再试一下安装compressed_image_transport插件 “sudo apt-get install ros-noetic-compressed-image-transport”
> 
> robot@LAPTOP-18QMGK0H：~$ sudo apt-get install ros-noetic-compressed-image-transport [sudo] 机器人密码：读取包列表...完成 构建依赖关系树 读取状态信息...完成 无法安装某些软件包。这可能意味着您请求了一个不可能的情况，或者如果您使用的是不稳定的发行版，则某些必需的软件包尚未创建或移出 Incoming。以下信息可能有助于解决这种情况：
> 
> 以下软件包有未满足的依赖关系： ros-noetic-compressed-image-transport ： 依赖： libturbojpeg0-dev 但不会安装 E： Unable to correct problems， you have held broken packages.安不上报错，老哥加个联系方式摄入交流一下吗？

我也都是看B站视频或者问AI解决的，你可以试着问下GPT

BruceYin888, 2025-03-12:
> 我也都是看B站视频或者问AI解决的，你可以试着问下GPT

哦哦，那我这个图像压缩安不上是什么原因？

Qiushui3, 2025-03-12:
> robot@LAPTOP-18QMGK0H：~$ sudo apt-get install ros-noetic-compressed-image-transport [sudo] 机器人密码：读取包列表...完成 构建依赖关系树 读取状态信息...完成 无法安装某些软件包。这可能意味着您请求了一个不可能的情况，或者如果您使用的是不稳定的发行版，则某些必需的软件包尚未创建或移出 [...]

你可以把报错信息发给GPT，它会给你很详细的操作过程

BruceYin888, 2025-03-12:
> 你可以把报错信息发给GPT，它会给你很详细的操作过程

我在电脑上复现了r3live这个slam框架，我就先用这个r3live了，解决这些问题太费时间了，太折磨了

BruceYin888, 2025-03-12:
> 你可以把报错信息发给GPT，它会给你很详细的操作过程

这个问题我知道什么原因，有个依赖我电脑上安装的版本高于它所需的版本，所以安装不上