hku-mars/FAST-LIVO2 #155: 图像与点云误匹配

shizhinian, 2025-03-11:
郑博你好，我在使用fast-livo2处理我自己数据集时出现了图像和点云误匹配的问题：

[image]
我的设备是有前后左右四个相机，将相应的外参修改后运行就出现了上图问题，想请问下可能的原因是什么呢？

shizhinian, 2025-03-11:
最后这个问题解决了，还是因为外参没设置正确，附上几张运行过程图，想问下郑博建图结果是否正常呢？

[image]
[image]
[image]
[image]
[image]
[image]
[image]

xuankuzcr, 2025-03-17:
可以的，看起来是正常的

shizhinian, 2025-03-18:
ok谢谢您了

closed 2025-03-18 by shizhinian