hku-mars/FAST-LIVO2 #157: using fastlivo2 with ouster lidar

techiemanvendra, 2025-03-12:
i am using ouster lidar and imu and running by on bag file data but i am able to map the area if i disabled th camera part but if i use camera also then it is not functioning here is my config file please @xuankuzcr @keleeven give some solution
common:
img_topic: "/camera/top_front/image_raw"
lid_topic: "/ouster/points"
imu_topic: "/ouster/imu"
img_en: 1
lidar_en: 1
ros_driver_bug_fix: false

extrin_calib:
extrinsic_T: [0.04165, 0.02326, -0.0284]
extrinsic_R: [1, 0, 0, 0, 1, 0, 0, 0, 1]
Rcl: [0.001,-0.027,-1.000,
-1.000,-0.001,-0.001,
-0.001,1.000,-0.027]
Pcl: [-0.021, -0.007, -0.116]

time_offset:
imu_time_offset: 0.0
img_time_offset: 0.1
exposure_time_init: 0.0

preprocess:
point_filter_num: 1
filter_size_surf: 0.1
lidar_type: 3 # Livox Avia LiDAR
scan_line: 32
blind: 1

vio:
max_iterations: 5
outlier_threshold: 1000 # 78 100 156 https://github.com/hku-mars/FAST-LIVO2/issues/100 200 500 700 infinite
img_point_cov: 100 # 100 1000
patch_size: 8
patch_pyrimid_level: 4
normal_en: true
raycast_en: false
inverse_composition_en: false
exposure_estimate_en: true
inv_expo_cov: 0.1

imu:
imu_en: true
imu_int_frame: 30
acc_cov: 0.5 # 0.2
gyr_cov: 0.3 # 0.5
b_acc_cov: 0.0001 # 0.1
b_gyr_cov: 0.0001 # 0.1

lio:
max_iterations: 5
dept_err: 0.02
beam_err: 0.05
min_eigen_value: 0.0025 # 0.005
voxel_size: 0.5
max_layer: 2
max_points_num: 50
layer_init_num: [5, 5, 5, 5, 5]

local_map:
map_sliding_en: false
half_map_size: 100
sliding_thresh: 8

uav:
imu_rate_odom: false
gravity_align_en: false

publish:
dense_map_en: true
pub_effect_point_en: false
pub_plane_en: false
pub_scan_num: 1
blind_rgb_points: 0.0

evo:
seq_name: "CBD_Building_01"
pose_output_en: false

pcd_save:
pcd_save_en: true
colmap_output_en: false # need to set interval = -1
filter_size_pcd: 0.15
interval: -1

camera_pinhole.yaml
cam_model: Pinhole
cam_width: 943
cam_height: 367
scale: 0.5
cam_fx: 248.85
cam_fy: 248.85
cam_cx: 471.5
cam_cy: 183.5
cam_d0: 0.0
cam_d1: 0.0
cam_d2: 0.0
cam_d3: 0.0

xuankuzcr, 2025-03-17:
Hello, could you please verify whether the timestamps of your camera and LiDAR are aligned on the same timeline?

techiemanvendra, 2025-03-17:
hey @xuankuzcr i am using some timestamp for camera and lidar