hku-mars/FAST-LIVO2 #160: 关于图像和点云存在异常缩放的现象。

gdd-DLMU, 2025-03-13:
大佬，你好：
我尝试使用了一个海上数据集，图像来自双目相机的左像头，点云来自ouster64(前雷达)。运行fast_livo2效果如下：

[image]

可以看到图片中的两个红框，a框和b框其实显示的是一样的，只是有缩放，如果a框放大一些应该就正常了。我设置了camra_pinhole.yaml中的scale参数，发现不起作用。
请教大佬这种情况是怎么回事呢，是相机和雷达的外参标定问题吗(Pcl和Pcl我是使用的你们开源的标定程序)？还是某个参数的设置。
如果能得到大佬百忙之中的指导，我将感激不尽。

gdd-DLMU, 2025-03-13:
当然，也非常欢迎其他大佬能够指出我的问题，感激不尽！

xuankuzcr, 2025-03-17:
The original dimensions of the camera should be filled in the camra_pinhole.yaml file. If this is correctly entered, then the issue is likely with the calibration of fx and fy.

gdd-DLMU, 2025-03-17:
> The original dimensions of the camera should be filled in the camra_pinhole.yaml file. If this is correctly entered, then the issue is likely with the calibration of fx and fy.

您好，非常感谢您一针见血指出我的问题。我在运行项目之前已经修改camra_pinhole.yaml中的参数，参数来源如下：

[image]

但是其中的cam_fy和cam_fx在原数据集中并未给出，我将图中的focal_length直接赋值给了cam_fy和cam_fx。
原数据集也未明确相机的具体型号和配置参数。
请问您有什么建议的方法能够标定cam_fy和cam_fx这两个参数嘛。
再次感谢！！！