hku-mars/FAST-LIVO2 #161: 使用fastlivo1中的数据集报错问题

nowPB, 2025-03-13:
修改了avia.yaml中的参数
`extrin_calib:
  Rcl: [0.00162756,-0.999991,0.00390957,
         -0.0126748,-0.00392989,-0.999912,
          0.999918,0.00157786,-0.012681]
  Pcl: [0.0409257, 0.0318424, -0.0927219]

time_offset: 
  img_time_offset: 0.0`
以及camera_pinhole.yaml中的参数为
`cam_model: Pinhole
cam_width: 640
cam_height: 512
# scale: 1.0
cam_fx: 431.795259219
cam_fy: 431.550090267
cam_cx: 310.833037316
cam_cy: 266.985989326
cam_d0: -0.0944205499243979
cam_d1: 0.0946727677776504
cam_d2: -0.00807970960613932
cam_d3: 8.07461209775283e-05`

播放包后出现以下报错信息：
`[ INFO] [1741864526.217055659]: Get LiDAR, its header time: 1577836979.199771
[ Preprocess ] Input point number: 24000 
[ Preprocess ] Output point number: 15988 
[ VIO ] Raw feature num: 15551
terminate called after throwing an instance of 'std::runtime_error'
  what():  Frame: provided image has not the same size as the camera model
[laserMapping-1] process has died [pid 42892, exit code -6, cmd /root/fastlivo2/fastlivo2_note/devel/lib/fast_livo/fastlivo_mapping __name:=laserMapping __log:=/root/.ros/log/f5932da2-fffb-11ef-a4af-107c6109993d/laserMapping-1.log].
log file: /root/.ros/log/f5932da2-fffb-11ef-a4af-107c6109993d/laserMapping-1*.log`
这是什么原因

xuankuzcr, 2025-03-17:
`camera_pinhole.yaml`中的参数设置有问题，`cam_width`和`cam_height`需要填入原始尺寸。

YyoungR, 2025-03-19:
你好，请问你解决了嘛，我把cam_width和cam_height换成1280和1024之后，跑出来的效果还是不如livo1，相机内参还是有问题感觉

YyoungR, 2025-03-19:
> `camera_pinhole.yaml`中的参数设置有问题，`cam_width`和`cam_height`需要填入原始尺寸。

郑博能否分享一下你测试LIVO1数据集的config，十分感谢，我把外参替换成了livo1的，内参替换成了livo1，并且cam_width`和`cam_height`填入原始尺寸1280、1024，然后对应的内参乘以了2，但是效果还是不行，很明显能看出来是内外参出的问题。

nowPB, 2025-03-25:
> > `camera_pinhole.yaml`中的参数设置有问题，`cam_width`和`cam_height`需要填入原始尺寸。
> 
> 郑博能否分享一下你测试LIVO1数据集的config，十分感谢，我把外参替换成了livo1的，内参替换成了livo1，并且cam_width`和`cam_height`填入原始尺寸1280、1024，然后对应的内参乘以了2，但是效果还是不行，很明显能看出来是内外参出的问题。

还要把img_time_offset设为0；scale设为0.5；不过着色效果看起来跟fastlivo1差不多，感觉还略微差点

zyllovelxx, 2025-03-25:
我在使用过程中采用了camera_pinhole.yaml
cam_model: Pinhole
cam_width: 640
cam_height: 512
scale: 2
cam_fx: 431.795259219
cam_fy: 431.550090267
cam_cx: 310.833037316
cam_cy: 266.985989326
cam_d0: -0.0944205499243979
cam_d1: 0.0946727677776504
cam_d2: -0.00807970960613932
cam_d3: 8.07461209775283e-05

img_time_offset设为0的复现效果如下
外参：
extrin_calib:
  extrinsic_T: [0.04165, 0.02326, -0.0284]
  extrinsic_R: [1, 0, 0, 0, 1, 0, 0, 0, 1]
  Rcl:  [0.00162756,-0.999991,0.00390957,
         -0.0126748,-0.00392989,-0.999912,
          0.999918,0.00157786,-0.012681] 
  Pcl: [0.0409257, 0.0318424, -0.0927219]

[image]