hku-mars/FAST-LIVO2 #165: 关于MID360的外参标定问题

darkyrs, 2025-03-14:
郑博您好，请问您使用过MID360进行测试吗，是使用的那种外参标定方法？
我是采用了基于目标的外参标定方法，使用的算法是[https://github.com/beltransen/velo2cam_calibration#](url)，考虑到MID360的非重复扫描特性，我对数据进行了累积以确保点云稠密，以1hz的频率发布数据，但是算法依然无法找到圆孔的圆心，请问针对MID360需要做哪些特殊的调整吗？谢谢

xuankuzcr, 2025-03-17:
我们最近搞了一套有目标的标定算法 (可用于Avia和Mid360)，用的板子和velo2cam的一样，是为了把camera和LiDAR的外参标定做得像“张正友标定法”一样简单。后续会开源。

[image]

YanhaoZhang, 2025-03-28:
您好 请问标定算法已经开源了嘛?

xuankuzcr, 2025-04-02:
还没有，开源会在此仓库更新

HonEYFIsh24, 2025-04-02:
我目前也在使用velo2cam_calibration，但是我目前遇到的问题是warmup阶段结束以后会持续报：
[WARNING] Invalid argument passed to canTransform argument source_frame in tf2 frame_ids cannot be empty
         at line 126 in /tmp/binarydeb/ros-noetic-tf2-0.7.7/src/buffer_core.cpp
目前我能确定的是MID360驱动运行后没有在tf中发布其frame，有可能是造成上述问题的原因。希望可以请教有遇到此类问题的朋友