hku-mars/FAST-LIVO2 #166: 求助

iorileslie, 2025-03-15:
郑伯，我测试时，在房间效果可以，从门口转到走廊/在走廊转向/感应灯突然亮起时会发生漂移，是因为距离墙面过近导致的吗？我应该修改哪些参数来优化？此外，私有数据集最终会开放吗（Narrow Corridor，Long Corridor ，Mining Tunnel ）

xuankuzcr, 2025-03-17:
距离太近avia的点会有特别大的畸变，可能需要blind一些点，然后打开raycasting模式

iorileslie, 2025-03-18:
> 距离太近avia的点会有特别大的畸变，可能需要blind一些点，然后打开raycasting模式

感谢回复，如果换成mid70近距离拍摄的效果会变好还是更差，mid70的盲区更近，但是我发现mid70的每秒点数少了很多

xuankuzcr, 2025-04-02:
从240k变成100k了，但是足够用了，要注意Mid70没内置IMU

iorileslie, 2025-05-15:
郑伯，我使用您的LIV_handhold方案同步，LIVO模式结果会出现明暗条纹，对着前方桌子拍，云台左右转动，结果点云中桌子左右会变宽，变宽部分上色是桌子后面物体的颜色；LIO模式下桌子宽度正常，rviz中按反射率上色显示也会有条纹。想问下两个模式下的条纹有关联吗，是时间同步问题导致这些问题的吗，雷达和相机时间对齐，具体是哪两个时刻对齐，

iorileslie, 2025-05-15:
郑伯，我mid70有同步的外置imu使用您的LIV_handhold方案同步，LIVO模式结果会出现明暗条纹，对着前方桌子拍，云台左右转动，结果点云中桌子左右会变宽，变宽部分上色是桌子后面物体的颜色；LIO模式下桌子宽度正常，rviz中按反射率上色显示也会有条纹。
想问下两个模式下的条纹有关联吗，是时间同步问题导致这些问题的吗，雷达和相机时间对齐，具体是哪两个时机的时刻对齐

TheShylorchicds, 2025-11-24:
> > 请问您的问题解决了吗？
> 
> 时间同步应该是主要原因，使用郑博的LIV_handhold同步方案后，图像相对雷达有个时间偏移，需要设置fastlivo2中的配置文件中img_time_offset: 0.1，我加上时间偏移后效果好了很多，剩下可能是其他问题，还在研究

您好，请问你使用mid70进行复现的话，雷达和外置imu这块是时间软件同步然后标定之后在使用fastlivo2算法的吗，这部分有点疑惑然后效果怎么样呢

iorileslie, 2025-11-25:
> 好的非常感谢，我当前使用的IMU只能软件同步，不知道最终效果如何，能方便提供一您使用的下IMU的型号吗？

hipnuc HI91