hku-mars/FAST-LIVO2 #167: 有没有用RK3588跑通这个算法的，我发现运行的时候不报错，就是运行一开始就里程计会飘，运行的官方数据集，x86能完美的运行

20211105, 2025-03-16:


prethn, 2025-03-17:
我们能跑啊，啥也不用改，opencv是ros noetic自带的和eigen我记得是3.3.7

prethn, 2025-03-17:
> 兄弟，方便问问你们具体是哪款3588吗，内存多大的
> 
> > 我们能跑啊，啥也不用改，opencv是ros noetic自带的和eigen我记得是3.3.7

用的是这款 ROC-RK3588S-PC 16G+128G ，当时找的教程 https://wiki.t-firefly.com/zh_CN/ROC-RK3588S-PC/started.html

20211105, 2025-03-18:
测试过程中发现，rk3588的这个板子，包的数据已经跑完了，建图程序还在跑，很卡的样子

20211105, 2025-03-18:
测试过程中发现，rk3588的这个板子，包的数据已经跑完了，建图程序还在跑，很卡的样子，我的也是16g+128g的，运行过程中内存是够的

gloryhry, 2025-03-25:
> 测试过程中发现，rk3588的这个板子，包的数据已经跑完了，建图程序还在跑，很卡的样子，我的也是16g+128g的，运行过程中内存是够的

这个问题一般是受限于计算性能的问题，单帧数据的处理时间过长，使得有大量数据累积到队列里了。可以降低一下雷达的单帧数据量试试。

xuankuzcr, 2025-04-02:
@20211105 `point_filter_num`设置成3或5，rviz只保留轨迹。Avia在RK3588S上大约30多毫秒一帧，详见这个工作 [FAST-LIVO2 on Resource-Constrained Platforms](https://arxiv.org/pdf/2501.13876)。

20211105, 2025-04-02:
@xuankuzcr 好勒，哥

asurasoft, 2025-05-22:
里程计漂移的问题解决了吗？