hku-mars/FAST-LIVO2 #174: point covariance calculation

biggiantpigeon, 2025-03-20:
In https://github.com/hku-mars/FAST-LIVO2/blob/main/src/voxel_map.cpp#L387, the point cov from lidar to world did not consider extrinsic from lidar to imu:
```
cov = state_.rot_end * cov * state_.rot_end.transpose() + (-point_crossmat) * rot_var * (-point_crossmat.transpose()) + t_var;
```
while in https://github.com/hku-mars/FAST-LIVO2/blob/6289d034f2754c073ff5c20f62bbdeaa5ff79bfd/src/voxel_map.cpp#L551 it does:
```
    var = (state_.rot_end * extR_) * var * (state_.rot_end * extR_).transpose() +
          (-point_crossmat) * state_.cov.block<3, 3>(0, 0) * (-point_crossmat).transpose() + state_.cov.block<3, 3>(3, 3);
```
Why is that? also some repeated operation (like calcBodyCov and transform of all points) here, seems to be of no need.

biggiantpigeon, 2025-03-21:
another question：in
https://github.com/hku-mars/FAST-LIVO2/blob/6289d034f2754c073ff5c20f62bbdeaa5ff79bfd/src/voxel_map.cpp#L470
why the delta state always this？ I thought after each iter, current state is updated and error-state should always be zero.

chaizheng2157, 2025-03-24:
> In https://github.com/hku-mars/FAST-LIVO2/blob/main/src/voxel_map.cpp#L387, the point cov from lidar to world did not consider extrinsic from lidar to imu:
> 
> ```
> cov = state_.rot_end * cov * state_.rot_end.transpose() + (-point_crossmat) * rot_var * (-point_crossmat.transpose()) + t_var;
> ```
> 
> while in
> 
> [FAST-LIVO2/src/voxel_map.cpp](https://github.com/hku-mars/FAST-LIVO2/blob/6289d034f2754c073ff5c20f62bbdeaa5ff79bfd/src/voxel_map.cpp#L551)
> 
> Line 551 in [6289d03](/hku-mars/FAST-LIVO2/commit/6289d034f2754c073ff5c20f62bbdeaa5ff79bfd)
> 
>  var = (state_.rot_end * extR_) * var * (state_.rot_end * extR_).transpose() + 
> it does:
> ```
>     var = (state_.rot_end * extR_) * var * (state_.rot_end * extR_).transpose() +
>           (-point_crossmat) * state_.cov.block<3, 3>(0, 0) * (-point_crossmat).transpose() + state_.cov.block<3, 3>(3, 3);
> ```
> 
> Why is that?

I have this question too, looking forward to the reply.

chaizheng2157, 2025-03-24:
> In https://github.com/hku-mars/FAST-LIVO2/blob/main/src/voxel_map.cpp#L387, the point cov from lidar to world did not consider extrinsic from lidar to imu:
>
> ```
> cov = state_.rot_end * cov * [...]

I have this problem too, and the coordinate of `point_crossmat` is different. Looking forward to the reply.

biggiantpigeon, 2025-03-26:
@chaizheng2157 I seem to get it. The one consider extrinsic, because point is in lidar coordinate. While the one does not (in LIO update) the point is in body (IMU) coordinate, which means extrinsic is applied to point before calc cov.