hku-mars/FAST-LIVO2 #176: Why is the PCD folder not generated in the Log folder？

Lockwuud, 2025-03-21:
Why is the PCD folder not generated in the Log folder and no .pcd files are created after enabling pcd_save_en?

common:
  img_topic: "/left_camera/image"
  lid_topic: "/livox/lidar_192_168_1_106"
  imu_topic: "/livox/imu_192_168_1_106"
  img_en: 0
  lidar_en: 1
  ros_driver_bug_fix: false

extrin_calib:
  extrinsic_T: [-0.011, -0.02329, 0.04412]
  extrinsic_R: [1, 0, 0, 0, 1, 0, 0, 0, 1]
  Rcl: [-0.00200, -0.99975, -0.02211,
      -0.00366, 0.02212, -0.99975,
       0.99999, -0.00192, -0.00371]
  Pcl: [0.00260, 0.05057, -0.00587]

time_offset: 
  imu_time_offset: 0.0
  img_time_offset: 0.1
  exposure_time_init: 0.0

preprocess: 
  point_filter_num: 2    #2 
  filter_size_surf: 0.1  # 0.1
  lidar_type: 1 # Livox Avia LiDAR
  scan_line: 6
  blind: 0.8

vio:
  max_iterations: 5
  outlier_threshold: 1000 # 78 100 156 #100 200 500 700 infinite
  img_point_cov: 100 # 100 1000 
  patch_size: 8
  patch_pyrimid_level: 4
  normal_en: true
  raycast_en: false
  inverse_composition_en: false
  exposure_estimate_en: true
  inv_expo_cov: 0.1

imu:
  imu_en: true
  imu_int_frame: 30
  acc_cov: 0.5 # 0.2
  gyr_cov: 0.3 # 0.5
  b_acc_cov: 0.0001 # 0.1
  b_gyr_cov: 0.0001 # 0.1

lio:
  max_iterations: 5
  dept_err: 0.02
  beam_err: 0.05
  min_eigen_value: 0.0025 # 0.005
  voxel_size: 0.5
  max_layer: 2
  max_points_num: 50
  layer_init_num: [5, 5, 5, 5, 5]

local_map:  
  map_sliding_en: false
  half_map_size: 100
  sliding_thresh: 8

uav:
  imu_rate_odom: false
  gravity_align_en: false

publish:
  dense_map_en: true
  pub_effect_point_en: false 
  pub_plane_en: false
  pub_scan_num: 1
  blind_rgb_points: 0.0

evo:
  # seq_name: "CBD_Building_01" 
  seq_name: "HKU_Campus"
  pose_output_en: false

pcd_save:
  pcd_save_en: true
  colmap_output_en: false # need to set interval = -1
  filter_size_pcd: 0.15
  interval: -1
  # how many LiDAR frames saved in each pcd file;
  # -1 : all frames will be saved in ONE pcd file, may lead to memory crash when having too much frames.

LaurenceLiry, 2025-03-26:
I also met this problem. I guess the reasons are as follows:

1. Workspace permission issues — unable to write files to your workspace. Try running:
`sudo chmod -R 777 ~/your_workspace/`

2. Modify line 95 in ~/your_workspace/src/FAST-LIVO2/src/LIVMapper.cpp, set pcd_save_en to true, then recompile and try ctrl+c when running the LIVO2.

Lockwuud, 2025-03-29:
Thank you very much. I followed your advice and pulled the latest code, which resolved this issue.