hku-mars/FAST-LIVO2 #177: time synchronization

wolichen, 2025-03-24:
Dr. Zheng, thank you for your excellent work.I’m just getting started with this.May I ask how you implement hardware time synchronization? I’ve searched for a long time but couldn’t find a detailed explanation. After I flashed the code, I found that the system couldn’t recognize the serial port. Then I checked online and saw that I need to configure the USB CDC virtual serial port using STM32CubeMX and flash it again. However, doing so would overwrite the previously flashed code, so the two projects need to be merged before flashing, but I’m not sure how to do that. Later, I also found that I could use a USB-to-TTL adapter, such as CH340. Could you please share how you implemented yours and guide me through it? Thank you so much!

ustc-flicar, 2025-03-24:
Here is another repository that Dr. Zheng has opened up:
https://github.com/xuankuzcr/LIV_handhold

Detailed instructions for hardware configuration and synchronization are included here

wolichen, 2025-03-24:
> Here is another repository that Dr. Zheng has opened up: https://github.com/xuankuzcr/LIV_handhold
> 
> Detailed instructions for hardware configuration and synchronization are included here

thank you!