hku-mars/FAST-LIVO2 #186: 相机时间是现在的，雷达和imu时间是2020年的

wolichen, 2025-03-25:
[ INFO] [1742884982.866229110]: Get image, its header time: 1742884982.860296
[ INFO] [1742884982.945780407]: Get LiDAR, its header time: 1590194850.899609
[ Preprocess ] Input point number: 24000 
[ Preprocess ] Output point number: 18618 
[ INFO] [1742884982.970876621]: Get image, its header time: 1742884982.960254
[ INFO] [1742884983.045415659]: Get LiDAR, its header time: 1590194850.999451
[ Preprocess ] Input point number: 24000 
[ Preprocess ] Output point number: 18565 
[ INFO] [1742884983.075863157]: Get image, its header time: 1742884983.060460
[ INFO] [1742884983.145710226]: Get LiDAR, its header time: 1590194851.099303
[ Preprocess ] Input point number: 24000 
[ Preprocess ] Output point number: 18204 
[ INFO] [1742884983.175289253]: Get image, its header time: 1742884983.160531
[ INFO] [1742884983.246013740]: Get LiDAR, its header time: 1590194851.199943
[ Preprocess ] Input point number: 24000 
[ Preprocess ] Output point number: 18623 
   这种情况应该怎么解决，谢谢各位大佬

ustc-flicar, 2025-03-25:
检查下lidar和imu的驱动程序

wolichen, 2025-03-25:
> 检查下lidar和imu的驱动程序

好的感谢，是检查sdk和ros驱动吗？ 我没有用外部gps模块，用的就用的fastllivo提供的时间同步代码进行了烧录，发送的仿真gprmc数据终端显示：Received RMC data packet: $GPRMC,000355.00,A,2237.496474,N,11356.089515,E,0.0,225.5,230520,2.3,W,A*2B  是要在代码里修改重新烧录吗

ustc-flicar, 2025-03-25:
进一步了解了，结合你在#117 issue中提问，你应该是follow着https://github.com/xuankuzcr/LIV_handhold的步骤，使用avia在复现硬件？

目前根据你pc接收到的gprmc和livox驱动的输出，驱动应该是已经正确根据gprmc打时间戳了(UTC 时间 2020年5月23日 00:03:55.00)

你可以尝试下把海康相机的驱动也换成https://github.com/xuankuzcr/LIV_handhold仓库中的[mvs_ros_pkg]试下，看看这个相机驱动有没有针对性做时间戳方面的调整

ustc-flicar, 2025-03-25:
进一步了解了，结合你在https://github.com/hku-mars/FAST-LIVO2/issues/177 issue中提问，你应该是follow着[https://github.com/xuankuzcr/LIV_handhold的步骤，使用avia在复现硬件？](https://github.com/xuankuzcr/LIV_handhold%E7%9A%84%E6%AD%A5%E9%AA%A4%EF%BC%8C%E4%BD%BF%E7%94%A8avia%E5%9C%A8%E5%A4%8D%E7%8E%B0%E7%A1%AC%E4%BB%B6%EF%BC%9F)

目前根据你pc接收到的gprmc和livox驱动的输出，驱动应该是已经正确根据gprmc打时间戳了(UTC 时间 2020年5月23日 00:03:55.00)

你可以尝试下把海康相机的驱动也换成[https://github.com/xuankuzcr/LIV_handhold仓库中的[mvs_ros_pkg]试下，看看这个相机驱动有没有针对性做时间戳方面的调整](https://github.com/xuankuzcr/LIV_handhold%E4%BB%93%E5%BA%93%E4%B8%AD%E7%9A%84%5Bmvs_ros_pkg%5D%E8%AF%95%E4%B8%8B%EF%BC%8C%E7%9C%8B%E7%9C%8B%E8%BF%99%E4%B8%AA%E7%9B%B8%E6%9C%BA%E9%A9%B1%E5%8A%A8%E6%9C%89%E6%B2%A1%E6%9C%89%E9%92%88%E5%AF%B9%E6%80%A7%E5%81%9A%E6%97%B6%E9%97%B4%E6%88%B3%E6%96%B9%E9%9D%A2%E7%9A%84%E8%B0%83%E6%95%B4)

ustc-flicar, 2025-03-25:
进一步了解了，结合你在https://github.com/hku-mars/FAST-LIVO2/issues/177 issue中提问，你应该是follow着https://github.com/xuankuzcr/LIV_handhold的步骤，使用avia在复现硬件？

目前根据你pc接收到的gprmc和livox驱动的输出，电脑通过串口成功接受gprmc信号且驱动应该是已经正确根据gprmc打时间戳了(UTC 时间 2020年5月23日 00:03:55.00)

你可以尝试下把海康相机的驱动也换成[https://github.com/xuankuzcr/LIV_handhold仓库中的[mvs_ros_pkg]试下，看看这个相机驱动有没有针对性做时间戳方面的调整

ustc-flicar, 2025-03-25:
进一步了解了，结合你在https://github.com/hku-mars/FAST-LIVO2/issues/177 issue中提问，你应该是follow着[https://github.com/xuankuzcr/LIV_handhold的步骤，使用avia在复现硬件？

目前根据你pc接收到的gprmc和livox驱动的输出，电脑通过串口成功接受gprmc信号且驱动应该是已经正确根据gprmc打时间戳了(UTC 时间 2020年5月23日 00:03:55.00)

你可以尝试下把海康相机的驱动也换成[https://github.com/xuankuzcr/LIV_handhold仓库中的[mvs_ros_pkg]试下，看看这个相机驱动有没有针对性做时间戳方面的调整

wolichen, 2025-03-25:
> 进一步了解了，结合你在[#177](https://github.com/hku-mars/FAST-LIVO2/issues/177) issue中提问，你应该是follow着https://github.com/xuankuzcr/LIV_handhold 的步骤，使用avia在复现硬件？
> 
> 目前根据你pc接收到的gprmc和livox驱动的输出，电脑通过串口成功接受gprmc信号且驱动应该是已经正确根据gprmc打时间戳了(UTC 时间 2020年5月23日 00:03:55.00)
> 
> 你可以尝试下把海康相机的驱动也换成https://github.com/xuankuzcr/LIV_handhold 仓库中的[mvs_ros_pkg]试下，看看这个相机驱动有没有针对性做时间戳方面的调整
非常感谢您的帮助，要不然我可能还会困在这里很长时间，通过您说的方法，我现在已经解决了这个问题。查看了下时间戳好像是以雷达的时间为主时间使相机时间对齐。现在时间戳没问题了，就是在移动过程中有时候会一直漂移。是我之前标定的结果有问题的原因吗，我是在时间戳没问题之前标定的，是因为这个原因吗。感谢！

ustc-flicar, 2025-03-25:
太好了！刚和郑纯然师兄也确认过了，整个系统是统一以STM32里面那个20年的时间戳做时钟源；

关于漂移，准确的外参标定确实至关重要；同时也可以先选择一些挑战性小的场景测试（如户外大空间多纹理场景，避免室内狭小空间、弱纹理等场景）

wolichen, 2025-03-25:
好的，非常感谢您和郑博的帮助！后面我重新标定下，再测试一些场景。

wolichen, 2025-03-25:
> 太好了！刚和郑纯然师兄也确认过了，整个系统是统一以STM32里面那个20年的时间戳做时钟源；
> 
> 关于漂移，准确的外参标定确实至关重要；同时也可以先选择一些挑战性小的场景测试（如户外大空间多纹理场景，避免室内狭小空间、弱纹理等场景）
您好，今天测试了一些场景发现这个漂移总是有，刚开始打开fastlivo2时设备静止时一切正常，有丰富特征点，如第一张图

[image]   但是拿起来前进或者左右晃时就会出现漂移，特征点越来越少直到没有，如图

[image]，很多场景下都是这样，您知道是怎么回事吗。感谢！
，

wolichen, 2025-03-25:
> 太好了！刚和郑纯然师兄也确认过了，整个系统是统一以STM32里面那个20年的时间戳做时钟源；
> 
> 关于漂移，准确的外参标定确实至关重要；同时也可以先选择一些挑战性小的场景测试（如户外大空间多纹理场景，避免室内狭小空间、弱纹理等场景）

您好，今天测试了一些场景发现这个漂移总是有，刚开始打开fastlivo2时设备静止时一切正常，有丰富特征点，如第一张图

[image]

但是拿起来前进或者左右晃时就会出现漂移，特征点越来越少直到没有，如图

[image]

很多场景下都是这样，您知道是怎么回事吗。感谢！

wolichen, 2025-03-25:
> 太好了！刚和郑纯然师兄也确认过了，整个系统是统一以STM32里面那个20年的时间戳做时钟源；
> 
> 关于漂移，准确的外参标定确实至关重要；同时也可以先选择一些挑战性小的场景测试（如户外大空间多纹理场景，避免室内狭小空间、弱纹理等场景）

您好，测试了一些场景发现这个漂移总是有，刚开始打开fastlivo2时设备静止时一切正常，有丰富特征点，如第一张图

[image]

但是拿起来前进或者左右晃时就会出现漂移，特征点越来越少直到没有，如图

[image]
测了几个场景都是这样，您知道是怎么回事吗。感谢！

ustc-flicar, 2025-03-25:
看起来外参似乎不太对，那个标定板投影在点云的板子上是一个扭曲的形状，然后有一部分还投到后面的墙上了

wolichen, 2025-03-26:
> 看起来外参似乎不太对，那个标定板投影在点云的板子上是一个扭曲的形状，然后有一部分还投到后面的墙上了，相机内参也检查下
> 
> [image]

quinnym233, 2025-08-06:
大佬好！我也遇到了这个问题，海康相机时间是现在的，雷达是20年的，但我用的就是https://github.com/xuankuzcr/LIV_handhold里面的相机驱动

quinnym233, 2025-08-06:
大佬好！我也遇到了这个问题，海康相机时间是现在的，雷达是20年的，雷达用的官方驱动，相机驱动用的https://github.com/xuankuzcr/LIV_handhold里面的，后面我把雷达驱动换成LIV_handhold里面的发现会报错，[ERROR] [1754474486.066064202]: open code: 28，请问有解决方法吗？
successfully set data type, handle: 2667686080, set_bit: 2
[2025-08-06 18:01:25.070] [console] [info]  Receive Ack: Id 256 Seq 8  [mid360_command_handler.cpp] [Handle] [64]
successfully set pattern mode, handle: 2667686080, set_bit: 0
[2025-08-06 18:01:25.070] [console] [info]  Receive Ack: Id 256 Seq 9  [mid360_command_handler.cpp] [Handle] [64]
successfully set lidar attitude, ip: 192.168.1.159
[2025-08-06 18:01:25.070] [console] [info]  Receive Ack: Id 256 Seq 10  [mid360_command_handler.cpp] [Handle] [64]
successfully change work mode, handle: 2667686080
[2025-08-06 18:01:25.070] [console] [info]  Receive Ack: Id 256 Seq 11  [mid360_command_handler.cpp] [Handle] [64]
successfully enable Livox Lidar imu, ip: 192.168.1.159
[2025-08-06 18:01:25.072] [console] [info]  Receive Command: Id 258 Seq 59496  [mid360_command_handler.cpp] [Handle] [67]
[2025-08-06 18:01:25.292] [console] [info]  Receive Command: Id 258 Seq 59502  [mid360_command_handler.cpp] [Handle] [67]
[INFO] [1754474486.065768709]: Support only one topic.
[ERROR] [1754474486.066064202]: open code: 28

[2025-08-06 18:01:26.066] [console] [info] Handle detection data, handle:2667686080, dev_type:9, sn:47MDN6E0030259, cmd_port:56100  [general_command_handler.cpp] [HandleDetectionData] [319]
[INFO] [1754474486.067451971]: Support only one topic.
[INFO] [1754474486.070782846]: livox/imu publish imu data, set ROS publisher queue size 256
[DEBUG] [1754474486.070866048]: Trying to publish message of type [sensor_msgs/Imu/6a62c6daae103f4ff57a132d6f95cec2] on a publisher with type [sensor_msgs/Imu/6a62c6daae103f4ff57a132d6f95cec2]

yccckid, 2026-08-24:
> 大佬好！我也遇到了这个问题，海康相机时间是现在的，雷达是20年的，雷达用的官方驱动，相机驱动用的[https://github.com/xuankuzcr/LIV_handhold里面的，后面我把雷达驱动换成LIV_handhold里面的发现会报错，[ERROR]](https://github.com/xuankuzcr/LIV_handhold%E9%87%8C%E9%9D%A2%E7%9A% [...]

> [ INFO] [1742884982.866229110]: Get image, its header time: 1742884982.860296 [ INFO] [1742884982.945780407]: Get LiDAR, its header time: 1590194850.899609 [ Preprocess ] Input point number: 24000 [ [...]

这个问题应该是相机没有读到生成的timeshare，ros默认当前的系统时间。