hku-mars/FAST-LIVO2 #189: 关于voxelmap 的协方差问题

ouguangjun, 2025-03-27:
感谢佬的工作，我想基于voxel map做一些开发，感觉fast-livo2的voxelmap比原版较完整。但我有些许疑问，可能会存在个人理解有误，希望作者和路过的广大网友能够解答。

1. 为什么计算 所有计算反对称矩阵的时候，都在前面增加负号？例如这段代码:
```
(-point_crossmat) * state_propagat.cov.block<3, 3>(0, 0) * (-point_crossmat).transpose()
```
这里两个负号感觉可以抵消掉，而且原论文也不存在负号。是考虑到什么方面的问题吗

2. 在voxel_map中的StateEstimation中为了在build_single_residual计算点面残差协方差，而计算的全局点协方差cov：
```
M3D cov = body_cov_list_[i];
M3D point_crossmat = cross_mat_list_[i];
cov = state_.rot_end * cov * state_.rot_end.transpose() + (-point_crossmat) * rot_var * (-point_crossmat.transpose()) + t_var;
```
cov第一项忽略了外参rot，第二项忽略了状态rot。

而StateEstimation 中计算kf更新部分，又根据build_single_residual返回的量 重新计算了点面残差协方差，而计算的全局点协方差变为:
```
var = state_propagat.rot_end * extR_ * ptpl_list_[i].body_cov_ * (state_propagat.rot_end * extR_).transpose();
```
这里少了 第二项和第三项。 

我的疑问是，是否在build_single_residual中计算的残差协方差可以直接用来计算卡尔曼滤波更新，而不需要重新计算。这两个版本的 全局点协方差都不同，我也看到注释有 理论上正确的版本:
```
// point_w cov (another_version)
// var = state_propagat.rot_end * extR_ * ptpl_list_[i].body_cov_ * (state_propagat.rot_end * extR_).transpose() +
//       state_propagat.cov.block<3, 3>(3, 3) - point_crossmat * state_propagat.cov.block<3, 3>(0, 0) * point_crossmat;
```
作者是否在测试的时候考虑到 精度 或者效率的不同？我看到很多基于voxelmap开发的都存在类似代码实现和论文不同的问题。

期待作者和网友的回复，可能存在个人理解有误，感谢！

biggiantpigeon, 2025-03-27:
about extrinsic, I just answered myself the same question in https://github.com/hku-mars/FAST-LIVO2/issues/174, it is because the point are in different coordinate.

about question2, you can refer to paper about the voxel map: Efficient and Probabilistic Adaptive Voxel Mapping for Accurate
Online LiDAR Odometry, it strictly follow equation(3) there.

ouguangjun, 2025-03-27:
> about extrinsic, I just answered myself the same question in [#174](https://github.com/hku-mars/FAST-LIVO2/issues/174), it is because the point are in different coordinate.
> 
> about question2, you can refer to paper about the voxel map: Efficient and Probabilistic Adaptive Voxel Mapping for Accurate Online LiDAR Odometry, it strictly follow equation(3) there.
> 
> However, I still can't work out the jacobian on the VIO part... hope some friend can show some hint on the derivation!

关于外参的说法，kf状态和map都是基于imu系的，那么激光雷达点转换到map系肯定需要外参，其测量协方差也需要外参的变换。

tanble, 2025-04-01:
激光和视觉slam中，协方差有哪些应用

xuankuzcr, 2025-04-02:
https://github.com/hku-mars/FAST-LIVO2/issues/89

closed 2025-04-03 by ouguangjun