hku-mars/FAST-LIVO2 #192: New Feature: Support for NTU-VIRAL Dataset

xuankuzcr, 2025-03-29:
🎉 We’ve added support for the NTU-VIRAL dataset! Here’s what’s included:
- Launch file and [unified parameters](https://github.com/hku-mars/FAST-LIVO2/blob/main/config/camera_NTU_VIRAL.yaml) for all sequences.
- A more user-friendly [evaluation method](https://github.com/hku-mars/FAST-LIVO2/tree/main/Log/result/ntu_viral) handling PRISM-IMU extrinsic parameters.
- RMSE improved from ~6cm (FAST-LIVO2 paper) to ~2cm, outperforming VIRAL-SLAM, BALM2, etc. (best result in SLAM systems to date).
- Runtime to data duration ratio is 1:6.

我们新增了对NTU-VIRAL数据集的支持！具体更新如下：
- 提供了Launch文件和适用于所有序列的[统一参数](https://github.com/hku-mars/FAST-LIVO2/blob/main/config/camera_NTU_VIRAL.yaml)。
- 提供了一套更友好的[评估方法](https://github.com/hku-mars/FAST-LIVO2/tree/main/Log/result/ntu_viral)，考虑了棱镜与IMU的外参。
- RMSE从约 6cm (如FAST-LIVO2论文中所述) 提升至约 2cm，优于VIRAL-SLAM和BALM2等 (是目前据我所知SLAM系统中的最佳结果）。
- 耗时与数据时长的比例为1:6。

| Scene  | Sequence | RMSE (cm) |
|--------|----------|-----------|
| eee    | 01       | 2.71      |
| eee    | 02       | 2.11      |
| eee    | 03       | 2.61      |
| nya    | 01       | 3.56      |
| nya    | 02       | 3.39      |
| nya    | 03       | 3.52      |
| sbs    | 01       | 2.34      |
| sbs    | 02       | 2.83      |
| sbs    | 03       | 3.11      |

xuankuzcr, 2025-03-29:
[image]

1391741823, 2025-03-30:
太厉害了郑博！附上论文里面其他模型的精度：（单位：m）

[image]

论文链接：
https://arxiv.org/pdf/2202.00379

NTU_VIRAL dataset链接：
https://ntu-aris.github.io/ntu_viral_dataset/

JingYCB, 2025-04-08:
> Hello and thank you for your excellent work. When running the data eee01.bag with the configuration file you provided, I can only achieve an accuracy of 2.4-2.6 dm after multiple attempts, how do I need to adjust it to achieve the best accuracy?

xuankuzcr, 2025-04-08:
Please follow the [evaluation guidance](https://github.com/hku-mars/FAST-LIVO2/blob/main/Log/result/ntu_viral).

JingYCB, 2025-04-09:
> Hello, you probably didn't consider the extrinsic conversion between IMU and PRISM coordinate systems before comparing the accuracy. Please refer to the author's link for details: https://github.com/hku-mars/FAST-LIVO2/blob/main/Log/result/ntu_viral/evaluate_viral.py
Thank you very much, resolved

JingYCB, 2025-04-09:
> [@JingYCB](https://github.com/JingYCB) Please follow the [evaluation guidance](https://github.com/hku-mars/FAST-LIVO2/blob/main/Log/result/ntu_viral).
Thank you very much, resolved

JingYCB, 2025-04-09:
When running fastlio2, do we also need to consider the extrinsic conversion between IMU and PRISM coordinate systems before comparing the accuracy? Further, under what circumstances do we need to convert to this coordinate system? Thank you very much.