hku-mars/FAST-LIVO2 #194: 相机和雷达标定

sr-001, 2025-03-31:
我购买的是fastlivo2项目中使用到的相机，但是在我使用livox_camera_calib这个工具标定时遇到了相机分辨率与要求分辨率不符的情况，我的相机截图保存下来是640x531，但是要求的是1920x1080，而这个相机最高分辨率是1280x1042，请问您是怎么使用这个库进行标定的呢？

julongwuyanzu, 2025-04-01:
解决了吗

xuankuzcr, 2025-04-02:
livox_camera_calib这个库没要求图像分辨率啊。另外 [mvs_ros_pkg](https://github.com/xuankuzcr/LIV_handhold/tree/main/mvs_ros_pkg) 里可以把image_scale设置成1.0，这样就得到原图了。

Es1erda, 2025-04-02:
> 我购买的是fastlivo2项目中使用到的相机，但是在我使用livox_camera_calib这个工具标定时遇到了相机分辨率与要求分辨率不符的情况，我的相机截图保存下来是640x531，但是要求的是1920x1080，而这个相机最高分辨率是1280x1042，请问您是怎么使用这个库进行标定的呢？

livox_camera_calib中应该对图像分辨率没有要求

sr-001, 2025-04-03:
但是点云数据和图片他的基础误差很大，就导致他的匹配是错误的

sr-001, 2025-04-03:
[图片]

sr-001, 2025-04-03:
[image]

sr-001, 2025-04-03:
[image]

sr-001, 2025-04-03:
> > 我购买的是fastlivo2项目中使用到的相机，但是在我使用livox_camera_calib这个工具标定时遇到了相机分辨率与要求分辨率不符的情况，我的相机截图保存下来是640x531，但是要求的是1920x1080，而这个相机最高分辨率是1280x1042，请问您是怎么使用这个库进行标定的呢？
> 
> livox_camera_calib中应该对图像分辨率没有要求

请问这个偏移我该怎么解决？

xuankuzcr, 2025-04-03:
用cad的外参作为初值

sr-001, 2025-04-04:
> 用cad的外参作为初值

谢谢，非常感谢