hku-mars/FAST-LIVO2 #199: Will FAST-LIVO 2 Work with Unitree 4D LiDAR L2?

Manivarashi, 2025-04-02:
I have recently bought Unitree 4D LiDAR L2 and it have in build IMU, i would like to FAST-LIVO2 with this LiDar, in Ubuntu 24.04 running ROS2 Jazzy?

RoboticEngineer-del, 2025-04-02:
Maybe you should ask this question in https://github.com/unitreerobotics

Manivarashi, 2025-04-02:
> Maybe you should ask this question in https://github.com/unitreerobotics

I did but considering Unitree 4d lidar is a spinning lidar and most of the FAST- LIVO 2 Demos run on Livox, Made me believe there might be a requirements from FAST-LIVO 2 on what lidar Can be used. SO I wanted to confirm?

xuankuzcr, 2025-04-03:
Both mechanical spinning and solid-state LiDAR perform well @Manivarashi

Manivarashi, 2025-04-03:
> Both mechanical spinning and solid-state LiDAR perform well [@Manivarashi](https://github.com/Manivarashi)

Understood, But how does it color. I mean, Lidar mentioned in demo is scanning forward. so Camera facing forward will colourise the  point cloud. But for a lidar that scan in 360 degrees, how can one camera colourise the entire point cloud.

Manivarashi, 2025-07-02:
hello, i have troubles with mending fastlivo2 sdk to work wtih unitree 4d lidar l2, can you provide a help?

mian-zhi, 2025-08-11:
Hi, did you successfully run on L2?