hku-mars/FAST-LIVO2 #202: 关于时间偏移参数含义及设定的问题（lidar_time_offset，img_time_offset）

JixinGao, 2025-04-03:
您好，感谢这么优秀的开源工作！

我不太理解config中img_time_offset: 0.1和 lidar_time_offset: -0.1的含义，因为看到您最新适配的ntu数据集也是增加了一个lidar的time offset，为什么需要时间偏移把接收的数据进行调整且偏移为什么是0.1/-0.1呢？我去掉这个时间偏移，在ntu数据集上进行测试，整体运行下来感觉也还可以，但精度会有一些损失。  

另外想问一下您，轨迹评估时好像时间有几处没对上，这是没关系的嘛？以下直接用您给的轨迹做的evo_ape ... -a -p
[image]
以下是我跑point-lio的结果
[image]

还麻烦您解答一下啦，感谢！

xuankuzcr, 2025-04-03:
感谢支持~

先解释`img_time_offset: 0.1`，在我们提供的[同步方案](https://github.com/xuankuzcr/LIV_handhold)中，LiDAR会将其时间戳存储到共享内存中，相机帧则会读取内存中最新的时间戳。由于LiDAR scan是一个段时间测量，需要等0.1秒扫完后才会将其header的时间戳存到内存里。此时被触发的相机读取的时间戳实际上是0.1秒前的，因此其真实采样时间戳需要+0.1s。

对于NTU-VIRAL数据集，我设置`lidar_time_offset: -0.1`，因为该数据集中的LiDAR scan提供的是帧末时间戳，而我们的系统需要的是header时间戳，因此需要-0.1s。这一判断是基于在静止条件下观察EKF估计出的位姿抖动程度得出的。

时间没对上不影响，只要在一个时间轴上就可以，evo会用高频的真值轨迹插值出LiDAR或相机采样时刻的位姿的。

JixinGao, 2025-04-03:
感谢解答！

closed 2025-04-09 by JixinGao