hku-mars/FAST-LIVO2 #205: 关于vio中误差的雅可比矩阵的求解。

18242360613, 2025-04-03:
`郑博士，你好。我在阅读vio.cpp源码时对视觉部分误差关于imu姿态雅可比矩阵的求解产生了疑问，如下所示：      
```
      V3D pf = Rcw * pt->pos_ + Pcw;
      V2D pc = cam->world2cam(pf);

      computeProjectionJacobian(pf, Jdpi);
      M3D p_hat;
      p_hat << SKEW_SYM_MATRX(pf);
......

```

18242360613, 2025-04-07:
我按照自己的理解编写了雅可比矩阵的求解公式，如下
`          Jimg << du, dv;
          Jimg = Jimg * state->inv_expo_time;
          Jimg = Jimg * inv_scale;
          V3D P_D(pt->pos_-Pwi);
          // std::cout<<"P_D: "<<P_D.transpose()<<std::endl;
          V3D tmp = Rwi.transpose()*P_D;
          // std::cout<<"tmp: "<<tmp.transpose()<<std::endl;
          M3D p_hat_new;
          p_hat_new << SKEW_SYM_MATRX(tmp);
          JdR_new = Jimg * Jdpi *Rci*p_hat_new;//旋转雅可比
          Jdt_new = -Jimg * Jdpi*Rci*Rwi.transpose();//平移雅可比
`
然后比较了原始代码的JdR，Jdt和我求出来的JdR_new以及 Jdt_new。

[image]

从数值上看两者之间还是有一些区别，最后在HIT_Graffiti_Wall_01.bag数据包上，结果有一些差异。上面是红色框是原始计算的结果，下面的绿框是修改后跑出来的结果。两者在高度方向上有一定的差异。我不清楚我求解的过程是否正确，原作者的代码我看不太懂。郑博士能否帮忙解释一下原始代码的求解过程。

[image]

closed 2025-04-17 by 18242360613

MingheWang, 2025-05-01:
我也有一样的疑问，残差应该是对系统的状态量Rcw求导求解过程中应该需要求解pf关于Rcw的导数，结果应该为-Rcw* (pt->pos_)^，不理解代码中为什么要对Rci求导。 @18242360613

biggiantpigeon, 2025-08-07:
Jacobian must be wrt Rwi and twi. But still I got different result as in code.
From my deduction, dI/dtwi part is same with author, but dI/dRwi is a little different.