hku-mars/FAST-LIVO2 #211: 请问是否有开源应用1中的无人机自动导航的代码呢？想具体了解到导航的应用

babyccc, 2025-04-08:
如题，想咨询将fast-livo2应用在导航里时配套的导航算法，整个系统如何实现

xuankuzcr, 2025-04-11:
硬件部分开源在[这里](https://github.com/xuankuzcr/LIV_handhold/tree/main/drone_cad)，自动导航算法流程如下，具体实现见论文。

[image]

babyccc, 2025-04-11:
> 硬件部分开源在[这里](https://github.com/xuankuzcr/LIV_handhold/tree/main/drone_cad)，自动导航算法流程如下，具体实现见论文。
> 
> [image]

感谢回复，请问论文有导航的具体实现吗？我看了您的论文和代码，似乎没有具体的导航实现

babyccc, 2025-04-11:
> > 硬件部分开源在[这里](https://github.com/xuankuzcr/LIV_handhold/tree/main/drone_cad)，自动导航算法流程如下，具体实现见论文。
> > [image]
> 
> 感谢回复，请问论文有导航的具体实现吗？我看了您的论文和代码，似乎没有具体的导航实现

请问郑博这里是只有自动导航算法流程，不开源相关代码吗？我很好奇fast-livo2怎么用到导航

WenxuanLi-whu, 2025-09-10:
怎么导航啊