hku-mars/FAST-LIVO2 #215: How to use external imu ?

lgh5054, 2025-04-10:
Hello, thank you for providing such an excellent open-source project.

Currently, we are trying to use an external IMU instead of Livox's built-in IMU in our algorithm. However, when running the code, the LiDAR’s base time operates with a timestamp from 2020, which causes a time synchronization issue with the external IMU, and it does not work properly. Is there any way to resolve this?

xuankuzcr, 2025-04-11:
In the current setup, the clock source is the STM32. If you intend to use an external IMU, you will need to configure the synchronization scheme according to the technical documentation of the IMU.

lgh5054, 2025-04-11:
@xuankuzcr 

Hello thank you for your response.

I am trying to use IMU data with MAVROS. The issue is that simply connecting Pixhawk telemetry to the STM32 does not guarantee that synchronization will work. If you have any experience with this or any reference code that I could look into, I would really appreciate it. I apologize for bothering you when you're busy.