hku-mars/FAST-LIVO2 #217: Imu Timestamp Jumps x seconds

biabia12, 2025-04-10:
Hi! First, I would like to thank you guys for your excellent work.

I am currently using a Livox Avia with a FLIR camera. I have used the open source [LIV Handhold](https://github.com/xuankuzcr/LIV_handhold.git) livox_ros_driver for the timestamp synchronization, and am using the same synchronization scheme presented in that repository via the STM32. I am using the connection scheme shown in [this diagram](https://github.com/xuankuzcr/LIV_handhold/issues/19#issuecomment-2196429412) using the livox converter 2.0 and the sync cable. For the camera driver, I have slightly modified the original camera driver of FLIR to incorporate the shared timestamps.

When I run the livox_ros_driver, camera driver, I have an issue occurring:
If I run the Lidar and Camera drivers and record a rosbag, then during playback after successful recording with the FAST-LIVO2 framework, the mapping randomly stops and the framework gives the error as shown in the picture:

[image]

As you can see, it says IMU Timestamp Jumps: x seconds, where the value is slowing increasing. Generally, the pattern seems like it is always increasing by either 0.005 or 0.0049 seconds. 
This error may occur randomly, or it may not occur at all sometimes. If I launch the framework directly after running my camera and lidar drivers, instead of recording a rosbag, then the same issue occurs. 

My questions:
1) Can anyone help with this? I'm having trouble figuring out where this issue lies and how to fix it. Any information would be appreciated.
2) Can I run the drivers and then run FAST LIVO2 directly on the data coming from the sensors, or is recording a rosbag necessary, since I noticed that in LIV handhold, a rosbag is used.

Apologies in advance if it is a simple mistake I am making. Thank you for your time!

xuankuzcr, 2025-04-11:
Have you considered the common grounding of the signal lines? Could you check your wiring connections as well as the frame header timestamps of the camera and LiDAR?

biabia12, 2025-04-12:
My signal lines are grounded properly as far as I can see. I checked my wiring. 
I checked the bagfile, as far as I can tell, the difference between IMU timestamps and the lidar timestamps is apparent such that, in my 23 second bagfile recording, it shows a 1 second time difference between timestamps at the end of the 23 second bagfile, which is not enough to cause such an issue?
On a look, fast-livo shows that the timestamp jump is usually close to the amount of second the bagfile has been played for.  Do you think the issue is due to the grounding or some sort of noise introduction in the wiring?

biabia12, 2025-04-18:
I found the issue, it was the sync cables ground and sync+ wire coming into contact with each other under the wires upper insulation. Thank you @xuankuzcr for pointing me in the right direction!

closed 2025-04-18 by biabia12