hku-mars/FAST-LIVO2 #218: 关于mid360的硬同步，GPRMC

fff-zx, 2025-04-12:
大佬，我参考着readme里面提到的 https://gitee.com/gwmunan/ros2/wikis/%E5%AE%9E%E6%88%98%E6%95%99%E7%A8%8B/Fast-LIVO%E5%A4%8D%E7%8E%B0%E4%B9%8B%E8%AE%BE%E5%A4%87%E9%85%8D%E7%BD%AE%E4%B8%8E%E7%BB%84%E8%A3%85 教程复现，设备是mid360和f103c8t6，mid360使用https://github.com/xuankuzcr/LIV_handhold/tree/main/livox_ros_driver2 提供的驱动，准确来说提这个issue的时候，我还并没有运行livox_ros_driver2。c8t6是直接烧录的https://github.com/xuankuzcr/LIV_handhold/tree/main/stm32_timersync-open ，接线如下

[image]
我直接将c8t6的GPRMC与PWM的GPIO引脚接到mid360的M12插头，但是我在打开LivoxViewer2 for Ubuntu v2.3.0查看时，发现了左上角的报错，如下

[image]
根据那个教程，我猜测可能是校验和的问题？请问一下，我是不是应该修改stm32的源码来解决?
因为该教程是FAST-LIVO1的，里面提到了使用mid360要修改stm32源码，但是我将他修改后的源码与https://github.com/xuankuzcr/LIV_handhold/tree/main/stm32_timersync-open 仔细对比发现是一模一样的，导致我现在定位不到问题在哪。求大佬解答帮助一下

fff-zx, 2025-04-13:
> 先不用管这个warning，运行下livox_ros_driver2，把LiDAR帧的时间戳打出来看看

我今天尝试了一下，同时运行的livox_ros_driver2和mvs_ros_driver,rosbag record录制下来之后用rqt_bag工具查看了一下，

[image]

[image]

这样子就算是对齐了吗?另外我发现在单独运行mvs_ros_driver的时候，相机topic数据的时间戳是固定不变的，这是对的吗

xuankuzcr, 2025-04-13:
是对的，看起来已经同步好了

RayRua, 2025-05-21:
请问一下硬同步后时间戳对齐是如何实现的

jiabotao, 2025-07-14:
@xuankuzcr @fff-zx 我用的雷达是mid-360,请问stm32只需要和雷达接两根线吗？我买的一分三线功能线给了4根，正负极供电把1，2，3，9引脚都用了。stm32不需要和mid360共地吗？