hku-mars/FAST-LIVO2 #220: How to implement time synchronization among GNSS boards, lidar, and cameras.

JACKLiuDay, 2025-04-14:
Dear Dr. Zheng,

I currently have a ZED-F9P GNSS board capable of outputting a PPS signal, along with a MID360 LiDAR and a Hikvision MV-CU013-A0UC camera. I have already achieved synchronization between the LiDAR and the camera using an STM32 development board and successfully reproduced FAST-LIVO2.

Now, I would like to use the GNSS board to synchronize the LiDAR and the camera. Do I still need the STM32 board? Can I directly connect the GNSS PPS signal to the trigger lines of the LiDAR and the camera to achieve my goal?

I would greatly appreciate your guidance on this matter.

Best regards

xuankuzcr, 2025-04-14:
More specifically, you can refer to this diagrammatic.

[image]

JACKLiuDay, 2025-04-14:
> More specifically, you can refer to this diagrammatic.
> 
> [image]

Thank you for your reply, Dr. Zheng. Does this image correspond to a published paper or an open-source synchronization method? It looks very similar to FAST-LIVO2's approach (also using STM32).

xuankuzcr, 2025-04-14:
Yeah, this is from MARS LVIG, an open-source project that integrates the FAST-LIVO2 system on the DJI M300 platform: https://mars.hku.hk/dataset.html. The [microchip program source code](https://drive.google.com/drive/folders/1yHp9SRchYPrKkvmGnrCPWdZB5u4GeIKb?usp=drive_link) is also publicly available.

JACKLiuDay, 2025-04-15:
Thank you for your reply, Dr. Zheng. I reviewed the LVIG project and synchronization code you provided, but I still have some questions. If I want to achieve synchronization between GNSS, LiDAR, and the camera, which specific board should I use? Is it the Arduino Uno board? And how should the wiring be implemented? It would be great if I could refer to your open-source FAST-LIVO hardware synchronization solution as a reference. https://github.com/xuankuzcr/LIV_handhold

JACKLiuDay, 2025-04-15:
> Yeah, this is from MARS LVIG, an open-source project that integrates the FAST-LIVO2 system on the DJI M300 platform: https://mars.hku.hk/dataset.html. The [microchip program source [...]
Thank you for your reply, Dr. Zheng. I reviewed the LVIG project and synchronization code you provided, but I still have some questions. If I want to achieve synchronization between GNSS, LiDAR, and the camera, which specific board should I use? Is it the Arduino Uno board? And how should the wiring be implemented? It would be great if I could refer to your open-source FAST-LIVO hardware synchronization solution as a reference. https://github.com/xuankuzcr/LIV_handhold

xuankuzcr, 2025-04-16:
Both Arduino and STM32 can be used. You can refer to the logic of the code written in Arduino to modify the STM32 code in [LIV_handhold](https://github.com/xuankuzcr/LIV_handhold). There isn't a ready-made STM32 version available.