hku-mars/FAST-LIVO2 #222: When I executed catkin_make, there were warnings and errors.

wangjiabaodante, 2025-04-14:
Built target vikit_ros
make: *** [Makefile:146: all] Error 2
Invoking "make -j8 -l8" failed
CMake Warning (dev) at /opt/ros/noetic/share/catkin/cmake/python.cmake:4 (find_package):
  Policy CMP0148 is not set: The FindPythonInterp and FindPythonLibs modules
  are removed.  Run "cmake --help-policy CMP0148" for policy details.  Use
  the cmake_policy command to set the policy and suppress this warning.

Call Stack (most recent call first):
  /opt/ros/noetic/share/catkin/cmake/all.cmake:164 (include)
  /opt/ros/noetic/share/catkin/cmake/catkinConfig.cmake:20 (include)
  CMakeLists.txt:58 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.8.10", minimum required is "3")
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Using Debian Python package layout
-- Using empy: /usr/lib/python3/dist-packages/em.py
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/dante/catkin_ws/build/test_results
-- Forcing gtest/gmock from source, though one was otherwise available.
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
CMake Deprecation Warning at /usr/src/googletest/CMakeLists.txt:4 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Deprecation Warning at /usr/src/googletest/googlemock/CMakeLists.txt:45 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Deprecation Warning at /usr/src/googletest/googletest/CMakeLists.txt:56 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Warning (dev) at /usr/src/googletest/googletest/cmake/internal_utils.cmake:249 (find_package):
  Policy CMP0148 is not set: The FindPythonInterp and FindPythonLibs modules
  are removed.  Run "cmake --help-policy CMP0148" for policy details.  Use
  the cmake_policy command to set the policy and suppress this warning.

Call Stack (most recent call first):
  /usr/src/googletest/googletest/CMakeLists.txt:91 (include)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found PythonInterp: /usr/bin/python3 (found version "3.8.10")
-- Using Python nosetests: /usr/bin/nosetests3
-- catkin 0.8.10
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 4 packages in topological order:
-- ~~  - vikit_common
-- ~~  - vikit_py
-- ~~  - vikit_ros
-- ~~  - fast_livo
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'vikit_common'
-- ==> add_subdirectory(rpg_vikit/vikit_common)
CMake Deprecation Warning at rpg_vikit/vikit_common/CMakeLists.txt:3 (CMAKE_MINIMUM_REQUIRED):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

Current CPU archtecture: x86_64
-- Found Eigen: /usr/local/include/eigen3
-- Eigen found (include: /usr/local/include/eigen3)
CMake Warning at /opt/ros/noetic/share/catkin/cmake/catkin_package.cmake:166 (message):
  catkin_package() DEPENDS on 'Sophus' but neither 'Sophus_INCLUDE_DIRS' nor
  'Sophus_LIBRARIES' is defined.
Call Stack (most recent call first):
  /opt/ros/noetic/share/catkin/cmake/catkin_package.cmake:102 (_catkin_package)
  rpg_vikit/vikit_common/CMakeLists.txt:46 (catkin_package)

-- +++ processing catkin package: 'vikit_py'
-- ==> add_subdirectory(rpg_vikit/vikit_py)
CMake Deprecation Warning at rpg_vikit/vikit_py/CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- +++ processing catkin package: 'vikit_ros'
-- ==> add_subdirectory(rpg_vikit/vikit_ros)
CMake Deprecation Warning at rpg_vikit/vikit_ros/CMakeLists.txt:3 (CMAKE_MINIMUM_REQUIRED):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
CMake Warning at /opt/ros/noetic/share/catkin/cmake/catkin_package.cmake:166 (message):
  catkin_package() DEPENDS on 'Sophus' but neither 'Sophus_INCLUDE_DIRS' nor
  'Sophus_LIBRARIES' is defined.
Call Stack (most recent call first):
  /opt/ros/noetic/share/catkin/cmake/catkin_package.cmake:102 (_catkin_package)
  rpg_vikit/vikit_ros/CMakeLists.txt:34 (catkin_package)

-- +++ processing catkin package: 'fast_livo'
-- ==> add_subdirectory(FAST-LIVO2)
CMake Deprecation Warning at FAST-LIVO2/CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Build Type: Release
-- Current CPU architecture: x86_64
-- Using general x86 optimizations: -O3 -march=native -mtune=native -funroll-loops
-- Processor count: 8
-- Multithreading enabled. Cores: 4
-- OpenMP found
-- mimalloc not found, proceeding without it
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
CMake Warning (dev) at /usr/lib/x86_64-linux-gnu/cmake/pcl/PCLConfig.cmake:152 (find_package):
  Policy CMP0144 is not set: find_package uses upper-case <PACKAGENAME>_ROOT
  variables.  Run "cmake --help-policy CMP0144" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.

  CMake variable EIGEN_ROOT is set to:

    /usr/include/eigen3

  For compatibility, find_package is ignoring the variable, but code in a
  .cmake module might still use it.
Call Stack (most recent call first):
  /usr/lib/x86_64-linux-gnu/cmake/pcl/PCLConfig.cmake:316 (find_eigen)
  /usr/lib/x86_64-linux-gnu/cmake/pcl/PCLConfig.cmake:551 (find_external_library)
  FAST-LIVO2/CMakeLists.txt:97 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found Eigen: /usr/local/include/eigen3 (Required is at least version "3.1")
-- Eigen found (include: /usr/local/include/eigen3, version: 3.4.0)
CMake Warning (dev) at /usr/lib/x86_64-linux-gnu/cmake/pcl/PCLConfig.cmake:134 (find_package):
  Policy CMP0167 is not set: The FindBoost module is removed.  Run "cmake
  --help-policy CMP0167" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

Call Stack (most recent call first):
  /usr/lib/x86_64-linux-gnu/cmake/pcl/PCLConfig.cmake:314 (find_boost)
  /usr/lib/x86_64-linux-gnu/cmake/pcl/PCLConfig.cmake:551 (find_external_library)
  FAST-LIVO2/CMakeLists.txt:97 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found Boost: /usr/include (found suitable version "1.71.0", minimum required is "1.55.0") found components: system filesystem date_time iostreams regex
CMake Warning (dev) at /usr/lib/x86_64-linux-gnu/cmake/pcl/Modules/FindFLANN.cmake:44 (find_package):
  Policy CMP0144 is not set: find_package uses upper-case <PACKAGENAME>_ROOT
  variables.  Run "cmake --help-policy CMP0144" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.

  CMake variable FLANN_ROOT is set to:

    /usr

asurasoft, 2025-05-07:
我也报错了，错误信息是

  5%] Building CXX object rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/src/performance_monitor.cpp.o
[  5%] Building CXX object rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/src/robust_cost.cpp.o
[  5%] Building CXX object rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/src/user_input_thread.cpp.o
[  5%] Building CXX object FAST-LIVO2/CMakeFiles/vio.dir/src/visual_point.cpp.o
[  6%] Building CXX object rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/src/pinhole_camera.cpp.o
[  6%] Building CXX object rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/src/equidistant_camera.cpp.o
[  6%] Building CXX object rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/src/polynomial_camera.cpp.o
[  6%] Building CXX object rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/src/homography.cpp.o
make[2]: *** No rule to make target '/home/wang/ros_t/Sophus/libSophus.so', needed by '/home/wang/catkin_ws/devel/lib/libvikit_common.so'.  Stop.
make[2]: *** Waiting for unfinished jobs....
[  6%] Building CXX object rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/src/img_align.cpp.o
[  6%] Built target _waterplus_map_tools_generate_messages_check_deps_GetWaypointByIndex
[  6%] Built target _waterplus_map_tools_generate_messages_check_deps_Waypoint
[  6%] Built target _waterplus_map_tools_generate_messages_check_deps_SaveWaypoints
[  6%] Built target _waterplus_map_tools_generate_messages_check_deps_AddNewWaypoint
[  6%] Built target _waterplus_map_tools_generate_messages_check_deps_GetChargerByName
[  6%] Built target _catkin_empty_exported_target
[  6%] Built target visualization_msgs_generate_messages_py
[  6%] Built target visualization_msgs_generate_messages_nodejs
[  6%] Built target visualization_msgs_generate_messages_eus
[  6%] Built target visualization_msgs_generate_messages_lisp
[  6%] Built target visualization_msgs_generate_messages_cpp
[  6%] Built target std_srvs_generate_messages_py
[  6%] Built target std_srvs_generate_messages_nodejs
[  6%] Built target std_srvs_generate_messages_eus
[  6%] Built target std_srvs_generate_messages_lisp
[  6%] Built target nav_msgs_generate_messages_py
[  6%] Linking CXX shared library /home/wang/catkin_ws/devel/lib/libvio.so
[  6%] Built target vio
make[1]: *** [CMakeFiles/Makefile2:4174: rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[  6%] Linking CXX shared library /home/wang/catkin_ws/devel/lib/liblaser_mapping.so
[  6%] Built target laser_mapping
make: *** [Makefile:141: all] Error 2
Invoking "make -j5 -l5" failed

raiots, 2025-05-09:
The same issue occurs during the catkin_make process:
```bash
(base) root@ubuntu20:~/catkin_ws# catkin_make -DPYTHON_EXECUTABLE=/usr/bin/python3
Base path: /root/catkin_ws
Source space: /root/catkin_ws/src
Build space: /root/catkin_ws/build
Devel space: /root/catkin_ws/devel
Install space: /root/catkin_ws/install
####
#### Running command: "cmake /root/catkin_ws/src -DPYTHON_EXECUTABLE=/usr/bin/python3 -DCATKIN_DEVEL_PREFIX=/root/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/root/catkin_ws/install -G Unix Makefiles" in "/root/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /root/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/noetic
-- This workspace overlays: /opt/ros/noetic
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.8.10", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Using Debian Python package layout
-- Found PY_em: /usr/lib/python3/dist-packages/em.py  
-- Using empy: /usr/lib/python3/dist-packages/em.py
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /root/catkin_ws/build/test_results
-- Forcing gtest/gmock from source, though one was otherwise available.
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python3 (found version "3.8.10") 
-- Found Threads: TRUE  
-- Using Python nosetests: /usr/bin/nosetests3
-- catkin 0.8.12
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 4 packages in topological order:
-- ~~  - vikit_common
-- ~~  - vikit_py
-- ~~  - vikit_ros
-- ~~  - fast_livo
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'vikit_common'
-- ==> add_subdirectory(rpg_vikit/vikit_common)
Current CPU archtecture: x86_64
-- Found OpenCV: /usr (found version "4.2.0") 
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'eigen3'
--   Found eigen3, version 3.3.7
-- Found Eigen: /usr/include/eigen3  
-- Eigen found (include: /usr/include/eigen3)
-- +++ processing catkin package: 'vikit_py'
-- ==> add_subdirectory(rpg_vikit/vikit_py)
-- +++ processing catkin package: 'vikit_ros'
-- ==> add_subdirectory(rpg_vikit/vikit_ros)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- +++ processing catkin package: 'fast_livo'
-- ==> add_subdirectory(FAST-LIVO2)
-- Build Type: Release
-- Current CPU architecture: x86_64
-- Using general x86 optimizations: -O3 -march=native -mtune=native -funroll-loops
-- Processor count: 8
-- Multithreading enabled. Cores: 4
-- OpenMP found
-- mimalloc not found, proceeding without it
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Found Eigen: /usr/include/eigen3 (Required is at least version "3.1") 
-- Eigen found (include: /usr/include/eigen3, version: 3.3.7)
-- Found Boost: /usr/include (found suitable version "1.71.0", minimum required is "1.55.0") found components: system filesystem date_time iostreams regex 
-- Checking for module 'flann'
--   Found flann, version 1.9.1
-- Found FLANN: /usr/lib/x86_64-linux-gnu/libflann_cpp.so  
-- The imported target "vtkParseOGLExt" references the file
   "/usr/bin/vtkParseOGLExt-7.1"
[... 123 lines cut ...]
[ 34%] Building CXX object rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/src/robust_cost.cpp.o
[ 37%] Building CXX object rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/src/user_input_thread.cpp.o
[ 40%] Building CXX object rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/src/pinhole_camera.cpp.o
[ 42%] Building CXX object rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/src/equidistant_camera.cpp.o
[ 45%] Building CXX object FAST-LIVO2/CMakeFiles/vio.dir/src/visual_point.cpp.o
[ 48%] Building CXX object rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/src/polynomial_camera.cpp.o
[ 51%] Building CXX object rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/src/homography.cpp.o
[ 54%] Linking CXX shared library /root/catkin_ws/devel/lib/libpre.so
[ 54%] Built target pre
make[2]: *** 没有规则可制作目标“/root/Sophus/build/libSophus.so”，由“/root/catkin_ws/devel/lib/libvikit_common.so” 需求。 停止。
make[2]: *** 正在等待未完成的任务....
[ 57%] Building CXX object rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/src/img_align.cpp.o
[ 60%] Linking CXX shared library /root/catkin_ws/devel/lib/libimu_proc.so
[ 60%] Built target imu_proc
make[1]: *** [CMakeFiles/Makefile2:651：rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/all] 错误 2
make[1]: *** 正在等待未完成的任务....
[ 62%] Linking CXX shared library /root/catkin_ws/devel/lib/libvio.so
[ 62%] Built target vio
[ 65%] Linking CXX shared library /root/catkin_ws/devel/lib/liblio.so
[ 65%] Built target lio
[ 68%] Linking CXX shared library /root/catkin_ws/devel/lib/liblaser_mapping.so
[ 68%] Built target laser_mapping
make: *** [Makefile:141：all] 错误 2
Invoking "make -j8 -l8" failed
```