hku-mars/FAST-LIVO2 #223: 能否就使用比较好的RGBD相机来运行FAST-LIVO2，就不用激光雷达来跑这个算法

smittenx, 2025-04-14:
郑博你好，如标题所示，我目前手里有一个质量还可以的RGBD相机，激光雷达因为不能进行硬同步，所以暂时没有办法满足这个算法的硬件需求，但我现在又比较想做下测试，所以我想就利用rgbd相机来实现这个算法，具体方法如下：
1. 不考虑实时或者bag包形式的运行
2. 该rgbd深度图和images是硬同步的，同时自带imu，所以我可以事先先将每一帧深度图转化为点云图，来代替硬同步的激光雷达，然后rgb相机就利用rgbd本身的rgb图像，通过修改算法的输入接口，将imu数据，rgbd点云数据以及rgb图像数据一起输入进入FAST-LIVO2算法中
3. 请问上面的方式是否有可行性，我现在比较担心的是mid360本身的fov和利用rgbd得到的点云fov差距较大，可能实际效果无法满足slam的要求

xuankuzcr, 2025-04-16:
fov不用担心，我担心的是你的rgbd中的rgb不是全局快门的，以及imu和rgb之间的时差，这俩保证没问题就能用

smittenx, 2025-04-16:
感谢回复，rgb是全局快门的，imu和rgb之间视差，我这边是计划把imu直接调成1000hz，然后不行就再加插值，来保证imu和rgb之时差，主要rgbd就只有近处是比较准的（2m %2），远处精度就不太能看了，这个我是想直接把远一点的点云直接滤波，但这块特征的数量就可能变少，感觉就稳定性就有点担心了。我这边现在去试试看，感谢郑博！！！

godfather1402, 2025-04-19:
Hi @smittenx , I also want to use an stereo camera with IMU integrated (realsense d457 or zed x mini) to run FAST-LIVO2. Let me know if you successfully run FAST LIVO2 with an RGBD camera

smittenx, 2025-04-22:
> Hi [@smittenx](https://github.com/smittenx) , I also want to use a stereo camera with IMU integrated (realsense d457 or zed x mini) to run FAST-LIVO2. Let me know if you successfully run FAST LIVO2 with an RGBD camera

@godfather1402 
https://github.com/hku-mars/FAST-LIVO2/issues/243 
I could run the process here, however, the final results were bad because my camera's IMU and RGB data streams aren't well time-aligned (synchronized). You need to verify if your IMU and RGB are synchronized.

zhangqi-sudo, 2025-07-30:
> fov不用担心，我担心的是你的rgbd中的rgb不是全局快门的，以及imu和rgb之间的时差，这俩保证没问题就能用

郑博好，请问如果只用相机不用激光的情况下，精度怎么样呢大概？