hku-mars/FAST-LIVO2 #224: voxel_map内存管理策略优化建议

letterso, 2025-04-15:
算法默认参数不启用map_sliding_en，在大范围场景下使用会占用很大内存。
启用则会出现两个问题（运行环境为orin nx）：

- 随着地图规模的增大，单次滑窗移除数据耗时会持续增加，实测会达到30-40ms或者以上。
- 通过范围移除点没有很好地控制大小，实测点的数量还是会持续增加。

改进建议：使用LRU策略对`voxel_map`进行管理，参考faster-lio实现

```cpp
  std::list<std::pair<VOXEL_LOCATION, VoxelOctoTree *>> voxel_map_cache; 
  std::unordered_map<VOXEL_LOCATION, typename std::list<std::pair<VOXEL_LOCATION, VoxelOctoTree *>>::iterator> voxel_map;
```
改进后，可以控制voxel_map内点的数量在指定范围内，实测单次处理最高耗时为8ms，平均耗时为1ms多点，精度基本没有变化，相对于原来无论是否map_sliding_en情况下，平均耗时都有少量优化，内存有明显减少，同时不会持续增加。

maolj-chn, 2025-04-15:
> * 测点的数量还是会持续增

请问您改进后的版本考虑开源吗

maolj-chn, 2025-04-15:
> 算法默认参数不启用map_sliding_en，在大范围场景下使用会占用很大内存。 启用则会出现两个问题（运行环境为orin nx）：
> 
> * 随着地图规模的增大，单次滑窗移除数据耗时会持续增加，实测会达到30-40ms或者以上。
> * 通过范围移除点没有很好地控制大小，实测点的数量还是会持续增加。
> 
> 改进建议：使用LRU策略对`voxel_map`进行管理，参考faster-lio实现
> 
>   std::list<std::pair<VOXEL_LOCATION, VoxelOctoTree *>> voxel_map_cache; 
>   std::unordered_map<VOXEL_LOCATION, typename std::list<std::pair<VOXEL_LOCATION, VoxelOctoTree *>>::iterator> voxel_map;
> 改进后，可以控制voxel_map内点的数量在指定范围内，实测单次处理最高耗时为8ms，平均耗时为1ms多点，精度基本没有变化，相对于原来无论是否map_sliding_en情况下，平均耗时都有少量优化，内存有明显减少，同时不会持续增加。

请问您改进后的版本考虑开源吗

xuankuzcr, 2025-04-16:
感谢 @letterso 的建议，这种方式确实更好。
内存方面：原来的 map sliding 是通过空间范围判断哪些体素可以删，清理效果并不精准；LRU 是基于访问频率进行管理，每插入一个新体素都淘汰一个最老的，总体素数量始终受控，其淘汰机制更精细。
耗时方面：原来的滑窗更新要遍历整个 voxel_map，看哪些体素超出滑窗范围，复杂度接近 $O(N)$；LRU 不需要做任何遍历，只要新加数据时看是否超过上限，直接删末尾节点即可。哈希+链表定位，查找和更新都是 $O(1)$。

我简写了下代码，大佬方便看下是这个意思么，后续我会多测些数据，评估精度影响和实测耗时，视情况将此策略合并到主分支。
```cpp
std::list<std::pair<VOXEL_LOCATION, VoxelOctoTree*>> voxel_map_cache_;
std::unordered_map<VOXEL_LOCATION, std::list<std::pair<VOXEL_LOCATION, VoxelOctoTree*>>::iterator> voxel_map_;
const size_t MAX_VOXEL_NUM = 20000;  // 可配置

void VoxelMapManager::UpdateVoxelMap(const std::vector<pointWithVar> &input_points)
{
  float voxel_size = config_setting_.max_voxel_size_;
  float planer_threshold = config_setting_.planner_threshold_;
  int max_layer = config_setting_.max_layer_;
  int max_points_num = config_setting_.max_points_num_;
  std::vector<int> layer_init_num = config_setting_.layer_init_num_;
  uint plsize = input_points.size();

  for (uint i = 0; i < plsize; i++)
  {
    const pointWithVar &p_v = input_points[i];
    float loc_xyz[3];
    for (int j = 0; j < 3; j++)
    {
      loc_xyz[j] = p_v.point_w[j] / voxel_size;
      if (loc_xyz[j] < 0) { loc_xyz[j] -= 1.0f; }
    }

    VOXEL_LOCATION position((int64_t)loc_xyz[0], (int64_t)loc_xyz[1], (int64_t)loc_xyz[2]);
    auto it = voxel_map_.find(position);

    if (it != voxel_map_.end())
    {
      // -------- 已存在体素：更新访问顺序 + 更新内容 --------
      voxel_map_cache_.splice(voxel_map_cache_.begin(), voxel_map_cache_, it->second);
      VoxelOctoTree* voxel = it->second->second;
      voxel->UpdateOctoTree(p_v);
    }
    else
    {
      // -------- 新体素：初始化并插入 --------
      VoxelOctoTree *octo_tree = new VoxelOctoTree(max_layer, 0, layer_init_num[0], max_points_num, planer_threshold);
      octo_tree->layer_init_num_ = layer_init_num;
      octo_tree->quater_length_ = voxel_size / 4.0f;
      octo_tree->voxel_center_[0] = (0.5f + position.x) * voxel_size;
      octo_tree->voxel_center_[1] = (0.5f + position.y) * voxel_size;
      octo_tree->voxel_center_[2] = (0.5f + position.z) * voxel_size;
      octo_tree->UpdateOctoTree(p_v);

      // 插入到前面（最近使用）
      voxel_map_cache_.emplace_front(position, octo_tree);

      // 更新哈希表
      voxel_map_[position] = voxel_map_cache_.begin();

      // -------- 检查是否超出体素上限 --------
      if (voxel_map_cache_.size() > MAX_VOXEL_NUM)
      {
        auto old_it = std::prev(voxel_map_cache_.end());

        // 删除最旧体素
        voxel_map_.erase(old_it->first);
        delete old_it->second;
        voxel_map_cache_.pop_back();
      }
    }
  }
}

```

letterso, 2025-04-17:
是这个意思，我在测试过程中发现，在遍历时进行voxel_map_的移除会影响效率，所以我这边是更新后统一删除，这是我的实现：
```cpp
void VoxelMapManager::UpdateVoxelMap(const std::vector<pointWithVar> &input_points)
{
  // 参数
  float voxel_size = config_setting_.max_voxel_size_;
  float planer_threshold = config_setting_.planner_threshold_;
  int max_layer = config_setting_.max_layer_;
  int max_points_num = config_setting_.max_points_num_;
  std::vector<int> layer_init_num = config_setting_.layer_init_num_;
  size_t plsize = input_points.size();

  // 遍历处理
  for (size_t i = 0; i < plsize; i++)
  {
    const pointWithVar p_v = input_points[i];
    // 计算voxel坐标
    float loc_xyz[3];
    for (int j = 0; j < 3; j++)
    {
      loc_xyz[j] = p_v.point_w[j] / voxel_size;
      if (loc_xyz[j] < 0) { loc_xyz[j] -= 1.0; }
    }
    VOXEL_LOCATION position((int64_t)loc_xyz[0], (int64_t)loc_xyz[1], (int64_t)loc_xyz[2]);
    auto iter = voxel_map_.find(position);
    // 如果点的位置已经存在voxel 那么更新
    if (iter != voxel_map_.end()) { 
      iter->second->second->UpdateOctoTree(p_v); 
      voxel_map_cache_.splice(voxel_map_cache_.begin(), voxel_map_cache_, iter->second); // 更新值并移动到头部
    }
    else
    {
      // 创建value
      VoxelOctoTree *octo_tree = new VoxelOctoTree(max_layer, 0, layer_init_num[0], max_points_num, planer_threshold);
      octo_tree->quater_length_ = voxel_size / 4;
      octo_tree->voxel_center_[0] = (0.5 + position.x) * voxel_size;
      octo_tree->voxel_center_[1] = (0.5 + position.y) * voxel_size;
      octo_tree->voxel_center_[2] = (0.5 + position.z) * voxel_size;
      octo_tree->temp_points_.push_back(p_v);
      octo_tree->new_points_++;
      octo_tree->layer_init_num_ = layer_init_num;

      // 插入新节点到头部
      voxel_map_cache_.emplace_front(position, octo_tree);
      voxel_map_.insert({position, voxel_map_cache_.begin()});
    }
  }

  // 容量检查，删除尾部节点
  if (voxel_map_cache_.size() >= CAPACITY)
  {
        while (voxel_map_cache_.size() >= CAPACITY)
        {
          delete voxel_map_cache_.back().second;
          auto last_key = voxel_map_cache_.back().first;
          voxel_map_.erase(last_key);
          voxel_map_cache_.pop_back();
        }
  }
}
```

xuankuzcr, 2025-04-17:
感谢分享！

hujiax380, 2025-04-18:
> [@xuankuzcr](https://github.com/xuankuzcr) 是这个意思，我在测试过程中发现，在遍历时进行voxel_map_的移除会影响效率，所以我这边是更新后统一删除，这是我的实现：
> 
> void VoxelMapManager::UpdateVoxelMap(const std::vector<pointWithVar> &input_points)
> {
>   // 参数
>   float voxel_size = config_setting_.max_voxel_size_;
>   float planer_threshold = config_setting_.planner_threshold_;
>   int max_layer = config_setting_.max_layer_;
>   int max_points_num = config_setting_.max_points_num_;
>   std::vector<int> layer_init_num = config_setting_.layer_init_num_;
>   size_t plsize = input_points.size();
> 
>   // 遍历处理
>   for (size_t i = 0; i < plsize; i++)
>   {
>     const pointWithVar p_v = input_points[i];
>     // 计算voxel坐标
>     float loc_xyz[3];
>     for (int j = 0; j < 3; j++)
>     {
>       loc_xyz[j] = p_v.point_w[j] / voxel_size;
>       if (loc_xyz[j] < 0) { loc_xyz[j] -= 1.0; }
>     }
>     VOXEL_LOCATION position((int64_t)loc_xyz[0], (int64_t)loc_xyz[1], (int64_t)loc_xyz[2]);
>     auto iter = voxel_map_.find(position);
>     // 如果点的位置已经存在voxel 那么更新
>     if (iter != voxel_map_.end()) { 
>       iter->second->second->UpdateOctoTree(p_v); 
>       voxel_map_cache_.splice(voxel_map_cache_.begin(), voxel_map_cache_, iter->second); // 更新值并移动到头部
>     }
>     else
>     {
>       // 创建value
>       VoxelOctoTree *octo_tree = new VoxelOctoTree(max_layer, 0, layer_init_num[0], max_points_num, planer_threshold);
>       octo_tree->quater_length_ = voxel_size / 4;
>       octo_tree->voxel_center_[0] = (0.5 + position.x) * voxel_size;
>       octo_tree->voxel_center_[1] = (0.5 + position.y) * voxel_size;
>       octo_tree->voxel_center_[2] = (0.5 + position.z) * voxel_size;
>       octo_tree->temp_points_.push_back(p_v);
>       octo_tree->new_points_++;
>       octo_tree->layer_init_num_ = layer_init_num;
> 
>       // 插入新节点到头部
>       voxel_map_cache_.emplace_front(position, octo_tree);
>       voxel_map_.insert({position, voxel_map_cache_.begin()});
>     }
>   }
> 
>   // 容量检查，删除尾部节点
>   if (voxel_map_cache_.size() >= CAPACITY)
>   {
>         while (voxel_map_cache_.size() >= CAPACITY)
>         {
>           delete voxel_map_cache_.back().second;
>           auto last_key = voxel_map_cache_.back().first;
>           voxel_map_.erase(last_key);
>           voxel_map_cache_.pop_back();
>         }
>   }
> }

请问大佬可以上传一份完整的项目代码吗

liweiwu6, 2025-05-12:
您好，在这里我想向您提出一个问题，就是在地图滑动里面，只对LIO里面的体素地图进行维护，为什么不需要对VIO中的feat_map进行适量的删除，我在代码中打印feat_map的大小，发现其体积一直在增大，可能是一个很简单的问题，但一直没有想明白，期待您的回复！

closed 2025-05-19 by letterso

67qi, 2025-06-18:
您好，想问一个问题，按照上面的方法加入了LRU机制，内存问题得以解决，但是在使用LIO模式建图时，地图相比于未加入LRU机制时产生了严重的漂移，请问有遇到过这种问题吗

iorileslie, 2025-07-11:
> > 您好，在这里我想向您提出一个问题，就是在地图滑动里面，只对LIO里面的体素地图进行维护，为什么不需要对VIO中的feat_map进行适量的删除，我在代码中打印feat_map的大小，发现其体积一直在增大，可能是一个很简单的问题，但一直没有想明白，期待您的回复！
> 
> feat_map我这边也加了LRU，内存也有明显减少，但VIO中类里面定义了feat_map，同时函数processFrame的参数里也有个feat_map，要优化一下再加LRU会安全点。

请问vio部分的LRU添加需要注意些什么,我按照 [https://github.com/hku-mars/FAST-LIVO2/issues/258](url) 的修改后,如果把vio的lru_size_设置的很小(400)时,当超过这个数时,程序在获取visual_submap->voxel_points[i]->obs_时会报错

zhengshunkai, 2025-11-10:
> 您好，想问一个问题，按照上面的方法加入了LRU机制，内存问题得以解决，但是在使用LIO模式建图时，地图相比于未加入LRU机制时产生了严重的漂移，请问有遇到过这种问题吗

你好，请问一下这个问题解决了吗？ 我也遇到了

G4419, 2026-01-29:
@zhengshunkai @67qi 你好，请问这个问题解决了吗

changbaohan, 2026-10-26:
> 12\. > > 您好，在这里我想向您提出一个问题，就是在地图滑动里面，只对LIO里面的体素地图进行维护，为什么不需要对VIO中的feat_map进行适量的删除，我在代码中打印feat_map的大小，发现其体积一直在增大，可能是一个很简单的问题，但一直没有想明白，期待您的回复！
>      > 
>      > 
>      > feat_map我这边也加了LRU，内存也有明显减少，但VIO中类里面定义了feat_map，同时函数processFrame的参数里也有个feat_map，要优化一下再加LRU会安全点。
>      
>      
>      请问vio部分的LRU添加需要注意些什么,我按照 [#258](https://github.com/hku-mars/FAST-LIVO2/issues/258)  的修改后,如果把vio的lru_size_设置的很小(400)时,当超过这个数时,程序在获取visual_submap->voxel_points[i]->obs_时会报错

请问解决了吗