hku-mars/FAST-LIVO2 #228: 关于自采数据生成彩色雷达点云质量的问题

wenhua00, 2025-04-16:
非常令人震撼的效果！

郑博你好，

我使用海康工业相机以及mid360对fast-livo2进行了复现，在很多场景都得到了很好的验证，但遇到了以下一些问题，想向郑博请教：

1. 我尝试在室内进行重建，生成彩色PCD如下图，
[image]
对地板铺设的泡沫垫重建效果较差（边缘信息不清晰）
[image]
获取的原始图片数据如下图，请问如何改进使得边缘锐化？
[image]

2.在路径“FAST-LIVO2/Log/Colmap/sparse/0”下的image.txt中，我注意到每个观测行都是“0.0 0.0 -1”；这是否表示该2D特征点没有关联到任何3D点，请问我应该如何修复这样的问题？

`# Image list with two lines of data per image:
#   IMAGE_ID, QW, QX, QY, QZ, TX, TY, TZ, CAMERA_ID, NAME
#   POINTS2D[] as (X, Y, POINT3D_ID)
1 0.506914 0.471686 -0.502669 0.517561 -0.001445 -0.054353 0.004106 1 00001.png
0.0 0.0 -1
2 0.507353 0.472474 -0.501970 0.517091 -0.000335 -0.053173 0.003773 1 00002.png
0.0 0.0 -1
3 0.508085 0.473990 -0.500360 0.516544 0.000509 -0.053282 0.004847 1 00003.png
0.0 0.0 -1
4 0.509783 0.473725 -0.500147 0.515319 0.002267 -0.053577 0.006994 1 00004.png
0.0 0.0 -1
5 0.512758 0.474589 -0.493789 0.517700 0.001028 -0.054647 0.009000 1 00005.png
0.0 0.0 -1
6 0.523451 0.466137 -0.484670 0.523265 0.002903 -0.058404 0.012474 1 00006.png
0.0 0.0 -1`

xuankuzcr, 2025-04-16:
感谢支持！
1. 看起来你采集的数据有问题，可以发给我下，我帮您定位下问题
2. #97

wenhua00, 2025-04-16:
郑博，数据发到您的邮箱里了，谢谢！

SpriteLin-ZJU, 2025-05-07:
The same issue, has it been resolved?