hku-mars/FAST-LIVO2 #230: 越界问题

zhengshunkai, 2025-04-16:
郑博好。我用自己数据跑了一下。金字塔层数选择3或4层，然后border的计算改成如下：
border = (patch_size_half + 1) * (1 << patch_pyrimid_level);
经常性的在如下代码的位置发生crash：
  float dv = 0.5f *
      ((w_ref_tl * img_ptr[scale * width] + w_ref_tr * img_ptr[scale + scale * width] +
        w_ref_bl * img_ptr[width * scale * 2] + w_ref_br * img_ptr[width * scale * 2 + scale]) -
       **_(w_ref_tl * img_ptr[-scale * width] + w_ref_tr * img_ptr[-scale * width + scale] + //这一行crash_**
        w_ref_bl * img_ptr[0] + w_ref_br * img_ptr[scale]));

(gdb) bt
#0  0x00007ffff42dd16b in VIOManager::updateState(cv::Mat, int) (this=0x555558db2710, img=..., level=<optimized out>)
    at vio.cpp:1612
#1  0x00007ffff42dfd25 in VIOManager::computeJacobianAndUpdateEKF(cv::Mat) (this=0x555558db2710, img=...)
    at vio.cpp:789
#2  0x00007ffff42e0128 in VIOManager::processFrame(cv::Mat&, std::vector<pointWithVar, std::allocator<pointWithVar> >&, std::unordered_map<VOXEL_LOCATION, std::shared_ptr<VoxelOctoTree>, std::hash<VOXEL_LOCATION>, std::equal_to<VOXEL_LOCATION>, std::allocator<std::pair<VOXEL_LOCATION const, std::shared_ptr<VoxelOctoTree> > > > const&, double)
    (this=0x555558db2710, img=..., pg=std::vector of length 2998, capacity 8052 = {...}, feat_map=std::unordered_map with 44446 elements = {...}, img_time=<optimized out>)
    at vio.cpp:1822

xuankuzcr, 2025-04-17:
如果只在你采的数据有问题，那应该是标定的参数有问题，你在算dv之前对img_ptr的索引加个保护吧

wssky99, 2025-04-19:
可以尝试在img_cbk中将存储部分改为img_buffer.push_back(img_cur.clone());，并在sync_packages中的case LIVO LIO部分修改为
`mtx_buffer.lock();
if(!img_buffer.empty())
{
  m.img = img_buffer.front().clone();
  img_buffer.pop_front();
  img_time_buffer.pop_front();
}    
mtx_buffer.unlock();`