hku-mars/FAST-LIVO2 #232: 画面累计延时和内存溢出问题

XiaoHa0914, 2025-04-17:
你好，郑博，我在运行fast-livo2算法时会有累计的画面延迟，伴随着里程计数据的延迟，并且十分钟左右内存就会达到100%并且程序报错停止，以下是我的参数配置，麻烦您看下：
common:
  img_topic: "/left_camera/image"
  lid_topic: "/livox/lidar"
  imu_topic: "/livox/imu"
  img_en: 1
  lidar_en: 1
  ros_driver_bug_fix: false

extrin_calib:
  extrinsic_T: [0.04165, 0.02326, -0.0284]
  extrinsic_R: [1, 0, 0, 0, 1, 0, 0, 0, 1]
#   Rcl: [0.00610193,-0.999863,-0.0154172,
#         -0.00615449,0.0153796,-0.999863,
#         0.999962,0.00619598,-0.0060598] 
#   Pcl: [0.0194384, 0.104689,-0.0251952]

  # Rcl: [-0.0120762,-0.999724,-0.0201273,
  #     -0.00531785,0.0201927,-0.999782,
  #     0.999913,-0.0119665,-0.00556023] 
  # Pcl: [0.0158474,0.0295131,0.101055]

  Rcl: [-0.00998343,-0.997413,-0.0711936,
     0.0203064,0.0709802,-0.997271,
     0.999744,-0.0114019,0.0195452,] 
  Pcl: [0.0279459,-0.425204,0.253204]

time_offset: 
  imu_time_offset: 0.0
  img_time_offset: 0.1
  exposure_time_init: 0.0

preprocess:
  point_filter_num: 1
  filter_size_surf: 0.1
  lidar_type: 1 # Livox Avia LiDAR
  scan_line: 4
  blind: 0.8

vio:
  max_iterations: 5
  outlier_threshold: 1000 # 78 100 156 #100 200 500 700 infinite
  img_point_cov: 100 # 100 1000 
  patch_size: 8
  patch_pyrimid_level: 4
  normal_en: true
  raycast_en: false
  inverse_composition_en: false
  exposure_estimate_en: true
  inv_expo_cov: 0.1

imu:
  imu_en: true
  imu_int_frame: 30
  acc_cov: 0.5 # 0.2
  gyr_cov: 0.3 # 0.5
  b_acc_cov: 0.0001 # 0.1
  b_gyr_cov: 0.0001 # 0.1

lio:
  max_iterations: 5
  dept_err: 0.02
  beam_err: 0.05
  min_eigen_value: 0.0025 # 0.005
  voxel_size: 0.5
  max_layer: 2
  max_points_num: 50
  layer_init_num: [5, 5, 5, 5, 5]

local_map:  
  map_sliding_en: false
  half_map_size: 100
  sliding_thresh: 8

uav:
  imu_rate_odom: false
  gravity_align_en: false

publish:
  dense_map_en: true
  pub_effect_point_en: false 
  pub_plane_en: false
  pub_scan_num: 1
  blind_rgb_points: 0.0

evo:
  seq_name: "CBD_Building_01"
  pose_output_en: false

pcd_save:
  pcd_save_en: false
  colmap_output_en: false # need to set interval = -1
  filter_size_pcd: 0.15
  interval: -1
  # how many LiDAR frames saved in each pcd file;
  # -1 : all frames will be saved in ONE pcd file, may lead to memory crash when having too much frames.

%YAML:1.0

#--------------------------------------------------------------------------------------------
# Camera Parameters. Adjust them!
#--------------------------------------------------------------------------------------------
# SerialNumber: "00G66398256"
TopicName: "left_camera/image"

TriggerEnable: 1 # 0 stands for Off, 1 stands for On

# ExposureAutoMode: 0 # 0 stands for Off, 1 stands for Once, 2 stands fr Continues
# ExposureTime: 5000 # uso

ExposureAutoMode: 2
AutoExposureTimeLower: 100
AutoExposureTimeUpper: 20000

image_scale: 1 # 1 0.5
GainAuto: 2 # Gain Auto, 0 stands for Off, 1 stands for Once, 2 stands for Continues
Gain: 15 # min: 0   max: 17.0166
Gamma: 0.7  # min: 0   max: 17.0166
GammaSelector: 1 # 0 stands for user, 1 stands for sRGB
# GammaEnable: 1

PixelFormat: 0 # 0: RGB8, 1: BayerRG8, 2: BayerRG12Packed, 3: BayerGB12Packed

xuankuzcr, 2025-04-17:
#225

XiaoHa0914, 2025-04-17:
已对相应参数进行修改
对于“在rviz中只保存最近10s的点云”这个解决方案，
是对下面这段段进行更改吗
- Alpha: 1
          Autocompute Intensity Bounds: false
          Autocompute Value Bounds:
            Max Value: 15
            Min Value: -5
            Value: false
          Axis: Z
          Channel Name: intensity
          Class: rviz/PointCloud2
          Color: 238; 238; 236
          Color Transformer: RGB8
          Decay Time: 10
          Enabled: true
          Invert Rainbow: false
          Max Color: 255; 255; 255
          Max Intensity: 159
          Min Color: 0; 0; 0
          Min Intensity: 5
          Name: surround
          Position Transformer: XYZ
          Queue Size: 100
          Selectable: false
          Size (Pixels): 1
          Size (m): 0.004999999888241291
          Style: Points
          Topic: /cloud_registered
          Unreliable: true
          Use Fixed Frame: true
          Use rainbow: true
          Value: true

XiaoHa0914, 2025-04-17:
你好，可以这样做吗
设置Decay Time: 10、Queue Size: 100