hku-mars/FAST-LIVO2 #234: Issue of execute with FusionPortableV2 data

DesWudio, 2025-04-18:
Dear team,

The FAST-LIVO2 had achieved really amazing mapping result!

I wourld like to run it with [FusionPortableV2](https://fusionportable.github.io/dataset/fusionportable_v2/) dataset which has mapping ground truth to take a look at the mapping performance.

Unfortunately, the RGB-colored point cloud result is very unclear, only the first few frames when the device is stationary got good result. Once the device begins to move, the RGB information mapping becomes distorted and inaccurate.

I have checked several times, but I still unable to fiind the source of the problem. Below are the parameters that I have modified

```
common:
  img_topic: "/left_camera/image"
  lid_topic: "/os_cloud_node/points"
  # imu_topic: "/os_cloud_node/imu/data_raw" # OS-IMU
  imu_topic: "/stim300/imu/data_raw" # Body IMU

extrin_calib:
  # FusionProtableV2 Handheld Sequences
  # # OSIMU
  # extrinsic_T: [0.006253, -0.011775, 0.007645]
  # extrinsic_R:
  #   [
  #     1.0000000,  0.0000000,  0.0000000,
  #     0.0000000,  1.0000000,  0.0000000,
  #     0.0000000,  0.0000000,  1.0000000 
  #   ]
  # BodyIMU
  # Translation of LiDAR frame w.r.t. IMU frame
  extrinsic_T: [ 0.035980, -0.017385, 0.033028 ]
  # Rotation matrix of LiDAR frame w.r.t. IMU frame 
  extrinsic_R:
    [
      0.9999586, 0.0083075, 0.0037154,
      -0.0083124, 0.9999646, 0.0013026,
      -0.0037045, -0.0013334, 0.9999923
    ]
  # Cam0
  # # Translation of LiDAR frame w.r.t. Camera frame
  Pcl: [ 0.0872, -0.0507, -0.0488 ]
  # Rotation matrix of LiDAR frame w.r.t. Camera frame
  Rcl:
    [
      0.0023380, -0.9998546, -0.0168944,
      0.0044804,  0.0169047, -0.9998471,
      0.9999872,  0.0022619,  0.0045193
    ]

time_offset: 
  img_time_offset: 0.06266

preprocess:
  # OS1-128
  point_filter_num: 3
  filter_size_surf: 0.1
  lidar_type: 3
  scan_line: 128
  blind: 0.5

vio:
  outlier_threshold: 200 # 78 100 156 #100 200 500 700 infinite

cam_model: Pinhole
cam_width: 1024
cam_height: 768
scale: 1.0
cam_fx: 602.4532
cam_fy: 602.4183
cam_cx: 520.8442
cam_cy: 397.5420
cam_d0: -0.0891
cam_d1: 0.0885
cam_d2: 0.0005
cam_d3: 0.0020
```

Could you please help me for this issue?

xuankuzcr, 2025-04-21:
I previously tested their dataset and found the data is not hard-synchronized. There are time discrepancies between the camera and LiDAR, and strangely, the time offsets vary across different frames. As a result, no matter how you set the `img_time_offset`, the effect is not satisfactory.

DesWudio, 2025-04-22:
Got it, thanks for providing this information