hku-mars/FAST-LIVO2 #237: z轴方向漂移问题

OYSJ1997, 2025-04-19:
郑博你好，我在运行自己的数据进行长距离建图时发现，将同一位置作为起点和终点，结果出现了漂移现象，我有关注到问题#84 ，我更改了下面的参数：acc_cov: 1.0，gyr_cov: 0.1，voxel_size: 1.0，但是误差仍然比较大，我采用的设备是hikon相机+avia雷达+硬同步方案，下面是的我的结果和具体参数：

xuankuzcr, 2025-04-21:
看起来点云质量有点差，同步怎么确定没问题的

OYSJ1997, 2025-04-21:
我是通过rqt_bag 观察rosbag中点云和图片的时间戳的差判断的，每帧点云和图片有大约0.1s的时间差，是否是因为同步的精度不够导致误差？

[image]

[image]

xuankuzcr, 2025-04-22:
把参数文件和数据发我下

OYSJ1997, 2025-04-22:
> 把参数文件和数据发我下
好的，下面是我的数据和配置文件，bag有点大，麻烦郑博您有时间帮我看一下，万分感谢！
链接: https://pan.baidu.com/s/1M04VRjWFr3OBLIDBnrl4Dw 提取码: kmrr 
--来自百度网盘超级会员v2的分享

xuankuzcr, 2025-04-22:
@OYSJ1997 仅凭这个数据，我无法确定是里程计的drift还是数据本身存在问题。但我有两个建议：

1. 启动时保持设备静止：建议静止超过5秒，最好达到10秒以上。
2. 在纹理丰富的场景验证外参和时差：方法是先保持设备静止，然后剧烈晃动，观察现象。

OYSJ1997, 2025-04-23:
> [@OYSJ1997](https://github.com/OYSJ1997) 仅凭这个数据，我无法确定是里程计的drift还是数据本身存在问题。但我有两个建议：
> 
> 1. 启动时保持设备静止：建议静止超过5秒，最好达到10秒以上。
> 2. 在纹理丰富的场景验证外参和时差：方法是先保持设备静止，然后剧烈晃动，观察现象。

好的，感谢您给出的建议，我再测试一下。