hku-mars/FAST-LIVO2 #238: 雷达-相机的外参标定，lidar_camera_calib

fff-zx, 2025-04-19:
您好，我今天在用lidar_camera_calib标定外参的时候，发现标定后效果反而更差，这里放两张图
[image]

[image]

是用的单数据标定calib.launch，png和pcd分别来着rqt_image_viewer截图，fast-lio2建图，为什么会出现这种情况呢，可以给个方向吗？我反复试过四五次都是越标定越差

另外，好像lidar_camera_calib不能用fast-livo2输出的pcd? 因为我看fast-livo2的pcd只包含xyz rgb，在运行launch的时候会报没有intensity的警告，但是fast-lio2的不会。

fff-zx, 2025-04-19:
我的雷达在我的相机正上方，理论上平移矩阵应该是[0，y，z]。
但是我查看标定输出的txt文件，里面的平移矩阵的x并不是0，而且离0很远（10cm左右）。
后面又打算尝试direct_visual_lidar_calibration这个标定包，但环境实在是难配置

fff-zx, 2025-04-21:
> > 我的雷达在我的相机正上方，理论上平移矩阵应该是[0，y，z]。 但是我查看标定输出的txt文件，里面的平移矩阵的x并不是0，而且离0很远（10cm左右）。 后面又打算尝试direct_visual_lidar_calibration这个标定包，但环境实在是难配置
> 
> dvlc可以使用docker。
> 建议赶紧试试郑博的新repo：https://github.com/hku-mars/FAST-Calib

感谢感谢

xuankuzcr, 2025-04-21:
Targetless calibration is both time-consuming and unstable. In comparison, FAST-Calib offers the following advantages:

1. **Faster:** It takes around 2 seconds, compared to the previous method which required over ten minutes.
2. **More Accurate:** It achieves pixel-level precision.
3. **Greater Stability:** It does not require an initial extrinsic guess and prevents negative optimization.
4. **Simplified Process for Mechanical LiDAR:** It can be performed using static data without needing to run FAST-LIO first.
5. **Lower Scene Requirements:** It only needs a target board within the FoV. In fact, targetless calibration demands stricter scene conditions.

fff-zx, 2025-04-21:
> Targetless calibration is both time-consuming and unstable. In comparison, FAST-Calib offers the following advantages:
>
> 1. **Faster:** It takes around 2 seconds, compared to the previous method [...]

感谢回复，这就去和老师沟通一下

fff-zx, 2025-05-15:
> > 我的雷达在我的相机正上方，理论上平移矩阵应该是[0，y，z]。 但是我查看标定输出的txt文件，里面的平移矩阵的x并不是0，而且离0很远（10cm左右）。 后面又打算尝试direct_visual_lidar_calibration这个标定包，但环境实在是难配置
> 
> dvlc可以使用docker。 建议赶紧试试郑博的新repo：https://github.com/hku-mars/FAST-Calib

老哥，我使用FAST-Calib标定的时候，还是越标定越差，标出来平移矩阵有2米高(实际也就10cm左右)。问过郑博他说1.horizon有畸变2.需要调整# Distance filter的xyz参数，老哥说可以直接在rviz里调整显示，直到把/filtered_cloud话题过滤到只剩下标定版，但是过滤这一步我不会，可以帮一下吗

Xujiaming1, 2025-05-15:
过滤是根据采集标定数据时LiDAR和标定版之间的相对位置来确定的，具体的LiDAR坐标系如下图所示：

[image]

[image]
以LiDAR原点为圆心，FAST-Calib工程中qr_params.yaml里的x_min、x_max表示筛选出在LiDAR前方多远的距离范围，这个距离应该把标定板包含在内同时尽可能减少包含除标定板外的点，对y和z同理。根据你的实际数据包情况作调整。
另外建议采集标定数据时LiDAR放置位置尽可能距离标定板2-3m左右。

Es1erda, 2025-05-15:
[image]