hku-mars/FAST-LIVO2 #250: 退化场景RGB点云输出的问题

letterso, 2025-04-24:
很感谢作者开源这么优秀的工作，关于RGB点云输出部分我这边觉得有些问题。

在`handleLIO`函数中，pcl_w_wait_pub 使用LIO估计状态变换到世界系，然后在`publish_frame_world`中赋颜色再输出。
```cpp
  for (int i = 0; i < size; i++) 
  {
    RGBpointBodyToWorld(&laserCloudFullRes->points[i], &laserCloudWorld->points[i]);
  }
  *pcl_w_wait_pub = *laserCloudWorld;
```

若场景存在退化`handleLIO`估计位姿存在误差时，`handleVIO`更新的位姿并不能用于修正点云在世界系下的坐标反而在`vio_manager->new_frame_->w2f(p_w)`中产生误差，应该保留原始点云信息，在`publish_frame_world`函数内直接投影到图像上获取颜色后，再使用`handleVIO`更新的位姿转换到世界系。

Lianghongnan, 2025-05-18:
就算保留原始body系的点云信息，你也需要LiDAR时刻的位姿比较准确，然后利用相机和LiDAR时刻的相对位姿以及外参才能进行投影。如果真出现了你说的LIO在一帧直接退化的情况，怎么投影都没用，但实际情况是`handleVIO`在不断修正位姿，视觉正常工作的情况下，你说的这种情况很难出现；视觉如果也退化了，那直接G了。关于更新地图，论文里说的很清楚，LiDAR位姿用来构建geometry，相机位姿用来render。

xuankuzcr, 2025-05-18:
@Lianghongnan 是这样的👍

letterso, 2025-05-19:
> 就算保留原始body系的点云信息，你也需要LiDAR时刻的位姿比较准确，然后利用相机和LiDAR时刻的相对位姿以及外参才能进行投影。如果真出现了你说的LIO在一帧直接退化的情况，怎么投影都没用，但实际情况是`handleVIO`在不断修正位姿，视觉正常工作的情况下，你说的这种情况很难出现；视觉如果也退化了，那直接G了。关于更新地图，论文里说的很清楚，LiDAR位姿用来构建geometry，相机位姿用来render。

感谢你的回复，我这边考虑的是激光雷达退化，视觉不退化的场景，若激光视觉都退化确实按照你的说法会G了。

按照更新流程，一般为LIO先更新，然后VIO再更新，当LIO退化后，VIO更新可以修正位姿，我这边的做法是使用VIO修正后的位姿替代LIO的位姿将点云帧转换到世界系，其余基本保持一致，实际修改后测试下来差别不大，后续我会采集多些激光退化相机不退化的场景进行测试，因为我现在是在做一些特定退化场景相关的工作，所以有些实现和想法和论文会有所出入，后续我会再仔细研读一下论文和代码，感谢你的建议。

closed 2025-05-19 by letterso

alin1229, 2026-09-21:
您好 您测试了更多退化场景以及阅读了代码之后觉得这个投影策略是否需要修改呢