hku-mars/FAST-LIVO2 #252: 🚀Exciting News: FAST-Calib, an Efficient and User-Friendly Calibration Tool, is Now Open-Sourced!

xuankuzcr, 2025-04-24:
🎉 We’ve recently developed an automatic target-based extrinsic calibration tool for LiDAR-camera systems! [Repository Link](https://github.com/hku-mars/FAST-Calib)

**Key highlights include:** 

- Achieve highly accurate calibration results **in just 2 seconds**.
- Support both solid-state and mechanical LiDAR.
- No need for any initial extrinsic parameters.

**In short, it makes extrinsic calibration as simple as intrinsic calibration!**

weepeter123, 2025-05-16:
非常优秀的工作！我使用了郑博开源的fast-calib这项工作来进行外参标定，发现外参的标定效率以及准确度都得到极大的提升。
下面分享几点我在使用时的经历和需要注意的事项：
       1、标定板的定制：我是在本地图文广告店定制的 (一般大一些的打印店都可以定制，我的是在赛可优数码印刷定制的)，到店以后跟店家提供需要制作的板子图片以及尺寸还有二维码的原图就可以，在这个网页 https://chev.me/arucogen/ 可以调整尺寸。一开始我想把板子做厚实一些，尽量减少变形带来的影响，选择了木塑板这种材质，标定效果并不好，我使用木塑板标定的外参平移结果大概从几十厘米到1米，基本无法使用，也找不到原因。后来经过郑博耐心调试和指导，发现这种材质对激光的反射率太低，表面会有糊的一堆杂乱点云。后来重新定制BT板 (表面pvc内部泡沫厚度5mm)，以及买了郑博推荐的白板架 https://e.tb.cn/h.6rLaIKxRMaEUZc6?tk=aLWcVisWr7Q 用来固定标定板，效果确实好了不少，但是可能是由于运输过程中板子在车里被弯曲，圆孔尺寸和二维码都有一定变形，对后续标定精度有一定影响。如果板子存在一定变形，可以试一下将marker size按照实际情况改小一些。我的marker实际尺寸为20cm，但板子有一定变形，我试着逐渐改小marker size数值，大概修改为18cm左右时计算得到的外参运行livo2赋色效果比较好。另外，要注意下marker size指的是黑色正方形边长，如果做了16cm或者18cm的，且没有形变，记得按照实际尺寸填写。上面提到的问题大家在做板子的时候尽量避免。

[image]

        2、算法参数调试方式和评估标准：在算法运行完毕时，可以用终端打印的 [Result] RMSE数值 以及rviz中点云上色的情况来评估外参标定的结果，RMSE一般要求在2cm以内认为标定效果可用，最好可以调节在1cm以内。主要说一下算法中Distance filter六个数值的设置方法，算法根据这六个数值剔除数值以外的雷达点云，过滤的目标是尽量只留下标定板的点云，每次运行算法后，可以在rviz中订阅/filtered_cloud话题，我一般会将/filtered_cloud这个话题改成醒目的颜色以及将它的size改为0.001，比较容易看清楚点云的情况，通过修改这六个数值剔除到这个话题的点云基本只有标定板，没有别的杂乱点云后，再将数值微调来使RMSE数值最小。改Distance filter参数时我是先将范围相对改大一些，先在视野中发现标定板的全部点云，再向内缩小范围 (需要注意雷达点云的坐标系，避免修改时改错方向)。运行标定算法时，还可以根据/edge_cloud话题和/center_z0_cloud来观察标定的圆心是否在四个孔的中心来进行再次确认标定效果，同时观察最后的彩色点云赋色效果也是比较重要的。

[image]

         3、一些注意点和我测试时效果较好的参数：标定环境尽量不要有强反光，如果实在不能避免，可以在地面布置一些毯子遮挡，标定板尽量离地，避免跟地面的点云粘连，不好修改滤波数值，我测试时相机雷达距离标定板3-5米左右。
最后，再次感谢郑博在我测试中给的耐心指导和帮助!!!

[image]

[image]

meng152634, 2026-01-16:
@weepeter123 @xuankuzcr 我最近也在使用FAST-Calib进行外参标定，标定结果RMSE在1cm以内，但是实际点云投影效果很差。我使用的标定板尺寸缩小为标准标定板的1/2了，请问标定板尺寸对结果有影响吗？标定板到相机的距离多远标定效果比较好呢？