hku-mars/FAST-LIVO2 #255: 将保存的文件应用于3DGS问题

wolichen, 2025-04-25:
感谢郑博及其团队和提问区各位大佬的帮助，我也慢慢的从0开始搭配硬件到现在用fastlivo2采集了好多场景。现在在我把相关文件应用到3DGS时又遇到了几个问题，希望各位大佬不吝赐教。
在avia.yaml开启相关设置后fastlivo2会保存以下全部文件

[image]，我把经过下采样的.pcd点云用cloudcompare转化为.ply文件将.ply文件图片和3个.txt文件（制成colmap格式）输入说点云文件有问题，后来发现只保留图片和3个.txt文件直接训练3dgs脚本会自动生成符合格式的.ply文件成功输出了结果，如图，此时相机分辨率是640x512的，

[image]

[image]

[image]，图一的轨迹是在果田一侧拍的，设备朝向几乎没有什么变化，没有进入垄间，图二是第一次训练时的结果，此时没有任何白色浮影，图三是经过多次训练后的结果（期间训练过同一地点进入垄间的数据），感觉好像树叶变清晰了，但是白色浮影也越来越多，都使用的同一个数据。

后来把相机分辨率改成1280x1024也重新进行标定了，进入垄间采了数据，轨迹和训练结果如图

[image]

[image]

[image]
图一是轨迹图，图二，三是3dgs训练的结果，图三是俯视图，结果非常不清晰而且白色浮影特别多，硬件就是livoxavia激光雷达，海康相机，stm32做时间同步，有懂的，做过的大佬可以指点下吗，谢谢！

closed 2025-04-28 by wolichen

VVivyy, 2025-10-30:
为什么我改了avia.yaml文件后运行不会保存images文件呢

timi-liuliang, 2026-01-06:
训练出的3DGS质量很差，不知道是不是转换后，像机位姿和点云不匹配，有训练出高质量的 3DGS的同学吗？

[image]