hku-mars/FAST-LIVO2 #260: 图像与点云匹配问题

HonEYFIsh24, 2025-04-27:
郑博您好，感谢您的杰出贡献！我最近在自己搭建硬件平台复现您的工作，我的激光雷达和相机的位置关系如下：

[image]
其中激光雷达使用了MID360，相机采用了海康威视的MV_CB016，完成了硬件触发以及内外参标定等。我将相机和激光雷达同向放置时取得了较好的效果，但为了日后拓展多相机的需求，我重新将设备做成了上图所示的结构，但在我使用它运行时遇到了如下图所示的问题（已重新经过内外参标定）：

[image]
其中雷达正前方的视角为：

[image]
可见此时将相机视角下的图像信息匹配到了激光雷达正前方的视角下。我想请教一下这种安装方式如果想要获得正确的建图效果是否需要修改配置或者做代码适配？

linyuxiu130, 2025-05-11:
您好，想问一下这个标定板是怎么做的呀

HonEYFIsh24, 2025-05-11:
需要按照GitHub页面提供的尺寸和二维码顺序自己画个图，注意二维码顺序不要搞错了。然后可以联系附近的图文店看看能不能做一下。

xuankuzcr, 2025-05-18:
要重新标外参，可以参考下#282 #280