hku-mars/FAST-LIVO2 #266: Failed to find match for field 'timestamp' when using Unilidar L1

Canadia123456, 2025-05-03:
I am using the Unilidar L1 instead of Livox Lidar (config file= https://github.com/unitreerobotics/point_lio_unilidar/blob/main/config/unilidar_l1.yaml) . I have correctly updated the avia.yaml file topics. However, when I run **roslaunch fast_livo mapping_unilidar.launch ** I get the following error:

Failed to find match for field 'timestamp'.
Failed to find match for field 'timestamp'.
Failed to find match for field 'timestamp'.
Failed to find match for field 'timestamp'.
Failed to find match for field 'timestamp'.
Failed to find match for field 'timestamp'.
Failed to find match for field 'timestamp'.

How do I easily fix this? 

unilidar_l1.yaml: 

                    common:
                        **lid_topic: "/unilidar/cloud" 
                        imu_topic: "/unilidar/imu"** 
                        con_frame: false # true: if you need to combine several LiDAR frames into one
                        con_frame_num: 1 # the number of frames combined
                        cut_frame: false # true: if you need to cut one LiDAR frame into several subframes
                        cut_frame_time_interval: 0.1 # should be integral fraction of 1 / LiDAR frequency   
                        time_lag_imu_to_lidar: 0.0 # Time offset between LiDAR and IMU calibrated by other algorithms, e.g., LI-Init (find in Readme)
                                                    # the timesample of IMU is transferred from the current timeline to LiDAR's timeline by subtracting this value
                    
                    preprocess:
                        **lidar_type: 5**
                        scan_line: 18
                        timestamp_unit: 0           # the unit of time/t field in the PointCloud2 rostopic: 0-second, 1-milisecond, 2-microsecond, 3-nanosecond.
                        blind: 0.5

Canadia123456, 2025-05-03:
I resolved this issue by selecting lidar_type: 1

However, I have come to realize that unilidar L1 is not compatible with Fast-Livo2 from this error message: 

[/rosout]
process[unitree_lidar_ros_node-2]: started with pid [4410]
[ERROR] [1746283384.952425641]: Client [/laserMapping] wants topic /unilidar/cloud to have datatype/md5sum [livox_ros_driver/CustomMsg/e4d6829bdfe657cb6c21a746c86b21a6], but our version has [sensor_msgs/PointCloud2/1158d486dd51d683ce2f1be655c3c181]. Dropping connection.

SimonovBlonsky, 2025-05-15:
> I resolved this issue by selecting lidar_type: 1
>
> However, I have come to realize that unilidar L1 is not compatible with Fast-Livo2 from this error message:
>
> [/rosout] [...]

The reason for that is the mismatch between sensor messages. Your lidar is publishing the PointCloud2 type while livox lidar uses its own custom message. Also, check the point fields in your PointCloud2 message (typically x,y,z,intensity,timestamp and ring), some times one of these names or datatype do not match the setting (e.g., the setting requires "t" for timestamp but your message has "times" instead).