hku-mars/FAST-LIVO2 #279: rviz中点云无法显示

Huashuijingying, 2025-05-13:
运行roslaunch fast_livo mapping_avia.launch并播放FAST-LIVO2-Dataset，点云地图无法正常显示，只能看到轨迹和相机，请问应该如何解决？

[image]

Huashuijingying, 2025-05-13:
```
zxy@ubuntu:~/catkin_ws$ roslaunch fast_livo mapping_avia.launch 
... logging to /home/zxy/.ros/log/a51ad1a0-2fac-11f0-bbc6-6b571b600683/roslaunch-ubuntu-7347.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:37989/

SUMMARY
========

PARAMETERS
 * /Pcl: [0.0194384, 0.104...
 * /Rcl: [0.00610193, -0.9...
 * /common/img_en: 1
 * /common/img_topic: /left_camera/image
 * /common/imu_topic: /livox/imu
 * /common/lid_topic: /livox/lidar
 * /common/lidar_en: 1
 * /common/ros_driver_bug_fix: False
 * /evo/pose_output_en: False
 * /evo/seq_name: CBD_Building_01
 * /extrin_calib/extrinsic_R: [1, 0, 0, 0, 1, 0...
 * /extrin_calib/extrinsic_T: [0.04165, 0.02326...
 * /imu/acc_cov: 0.5
 * /imu/b_acc_cov: 0.0001
 * /imu/b_gyr_cov: 0.0001
 * /imu/gyr_cov: 0.3
 * /imu/imu_en: True
 * /imu/imu_int_frame: 30
 * /laserMapping/cam_cx: 626.91359
 * /laserMapping/cam_cy: 522.799224
 * /laserMapping/cam_d0: -0.07616
 * /laserMapping/cam_d1: 0.123001
 * /laserMapping/cam_d2: -0.00113
 * /laserMapping/cam_d3: 0.000251
 * /laserMapping/cam_fx: 1293.56944
 * /laserMapping/cam_fy: 1293.3155
 * /laserMapping/cam_height: 1024
 * /laserMapping/cam_model: Pinhole
 * /laserMapping/cam_width: 1280
 * /laserMapping/scale: 0.5
 * /lio/beam_err: 0.05
 * /lio/dept_err: 0.02
 * /lio/layer_init_num: [5, 5, 5, 5, 5]
 * /lio/max_iterations: 5
 * /lio/max_layer: 2
 * /lio/max_points_num: 50
 * /lio/min_eigen_value: 0.0025
 * /lio/voxel_size: 0.5
 * /local_map/half_map_size: 100
 * /local_map/map_sliding_en: False
 * /local_map/sliding_thresh: 8
 * /pcd_save/colmap_output_en: False
 * /pcd_save/filter_size_pcd: 0.15
 * /pcd_save/interval: -1
 * /pcd_save/pcd_save_en: False
 * /preprocess/blind: 0.8
 * /preprocess/filter_size_surf: 0.1
 * /preprocess/lidar_type: 1
 * /preprocess/point_filter_num: 1
 * /preprocess/scan_line: 6
 * /publish/blind_rgb_points: 0.0
 * /publish/dense_map_en: True
 * /publish/pub_effect_point_en: False
 * /publish/pub_plane_en: False
 * /publish/pub_scan_num: 1
 * /rosdistro: noetic
 * /rosversion: 1.17.2
[... 92 lines cut ...]
+-------------------------------------------------------------+
|                         VIO Time                            |
+-------------------------------------------------------------+
| Sparse Map Size               | 1234                        |
+-------------------------------------------------------------+
| Algorithm Stage               | Time (secs)                 |
+-------------------------------------------------------------+
| retrieveFromVisualSparseMap   | 0.000001                    |
| computeJacobianAndUpdateEKF   | 0.000001                    |
| -> computeJacobian            | 0.000000                    |
| -> updateEKF                  | 0.000000                    |
| generateVisualMapPoints       | 0.000381                    |
| updateVisualMapPoints         | 0.000001                    |
| updateReferencePatch          | 0.000001                    |
+-------------------------------------------------------------+
| Current Total Time            | 0.000385                    |
| Average Total Time            | 0.000385                    |
+-------------------------------------------------------------+
[ LIO ] Raw feature num: 20776, downsampled feature num:10723 effective feature num: 8842 average residual: 0.0349627
[ LIO ] Raw feature num: 20776, downsampled feature num:10723 effective feature num: 8938 average residual: 0.0222045
[ LIO ] Raw feature num: 20776, downsampled feature num:10723 effective feature num: 8935 average residual: 0.0220713
[ LIO ] Raw feature num: 20776, downsampled feature num:10723 effective feature num: 8936 average residual: 0.0220945
[ LIO ] Raw feature num: 20776, downsampled feature num:10723 effective feature num: 8936 average residual: 0.0220979
[ LIO ] Update Voxel Map
```

fff-zx, 2025-05-17:
> 运行roslaunch fast_livo mapping_avia.launch并播放FAST-LIVO2-Dataset，点云地图无法正常显示，只能看到轨迹和相机，请问应该如何解决？
> 
> [image]

我也遇到了一样的问题，请问你是在播放rosbag的时候遇到的吗？

FelixCooper1026, 2025-05-29:
你是虚拟机吗，我是Vmware Ubuntu 20.04 虚拟机，也是只有轨迹和黑白（灰度）的点云，没有赋色。

Huashuijingying, 2025-05-29:
> 你是虚拟机吗，我是Vmware Ubuntu 20.04 虚拟机，也是只有轨迹和黑白（灰度）的点云，没有赋色。

Vmware Ubuntu 20.04的虚拟机和jetsonNX都遇到这个问题

ovoayu, 2025-06-27:
> > 运行roslaunch fast_livo mapping_avia.launch并播放FAST-LIVO2-Dataset，点云地图无法正常显示，只能看到轨迹和相机，请问应该如何解决？
> > [image]
> 
> 我也遇到了一样的问题，请问你是在播放rosbag的时候遇到的吗？

我也遇到相同的问题，请问解决了吗

CavaPhPy, 2025-08-03:
目前找到的解决方案有2个（二选一）：
1、虚拟机设置——显示——加速3D图形。把这个选项取消勾选，即取消加速；
2、在ubuntu系统中，打开终端，设置使用软件计算。命令：export LIBGL_ALWAYS_SOFTWARE=1

根本原因我不清楚，按照搜索的结果，可能是 VMware 的 3D 加速驱动和 Ubuntu 20.04 的 Mesa 驱动之间的兼容性有关。

如果需要硬件计算的话，可能只有安装到物理机上才能解决。 如果有更好的解决方案，欢迎@我。